Arduino-IRremote/Arduino-IRremote #387: Write Tutorials and Make Videos

z3t0, 2016-12-04:
Need to create up to date tutorials and videos explaining this library and how to use it.

**Details**

- [ ] How to use Github Issues
- [ ] How to install
- [ ] How to use the docs
- [ ] How to do more advanced stuff, eg. modifying the library, multi receiver 
- [ ] How to find more help, report issues, contact etc.
- [ ] Update website

#386 @z3t0

bengtmartensson, 2016-12-04:
How 'bout some sort of discussion forum? Issues are issues, and Gitter simple does not do it.

> How to use Github Issues

The seniors already know (project independent!) how to write issues; there will always be rookies who cannot tell the difference between help requests, bugs, and enhancement requests; learn to live with it :-).

z3t0, 2016-12-04:
@bengtmartensson That's a really good point that I did not really think about at all. How about something like [this](https://github.com/discourse/discourse0

AnalysIR, 2016-12-04:
@z3t0  The link is broken....[but this may be more appropriate](https://blog.discourse.org/2016/03/free-discourse-forum-hosting-for-community-friendly-github-projects/)

There are some conditions for free hosting!

Otherwise there is usually a fee for 'discourse' hosting (uses more resources than older systems).

z3t0, 2016-12-04:
@AnalysIR fixed the link, I think we may be talking about the same thing

z3t0, 2016-12-04:
@AnalysIR it seems we don't really qualify for the free hosting as we dont have 2000+ stars or 30+ contributors hmm

z3t0, 2016-12-04:
Would it be realistic to start a crowd source for a fund that would allow us to host a forum? If so maybe we should find some cheap options to look at

yesyesuk, 2016-12-05:
I have a dedicated webserver that is not doing all that much. It should be able to run a forum if it doesn't have too much traffic. I'd be willing to host it as my thanks for the library. ;) However, I'm probably lacking skills to troubleshoot if anything went wrong with it. So I can provide the hosting but it would be good if someone else was able to be the forum admin.

I'm currently on vacation for another week. We can discuss in more detail when I'm back, if you are interested. Please prod me in case i forget. ;)

Chris

z3t0, 2016-12-05:
@yesyesuk Thanks for the contribution! I think we might be able to try that as long as the others are okay with it. I should be able to help set up and admin it as I have a server at home as well but for some reason my ISP insists on not allowing me to port forward...

z3t0, 2016-12-05:
As another thought why not just do something on the Arduino Forums?

closed 2020-10-17 by ArminJo