Arduino-IRremote/Arduino-IRremote #389: Library Goals Statement

z3t0, 2016-12-04:
Need to create some sort of goal statement so that it is clear what the intent of this library is. This is mainly to distinguish between what is and is not within the scope of this library/

#386 @z3t0

z3t0, 2017-04-18:
This might be a good discussion to have before we move forward with some of the major changes in #437 such as soft PWM support.

We need to ask ourselves:
- What is the target audience of this library?
- What problems does it aim to solve?
- What problems does it leave to the user to figure out (i.e. more advanced use cases)

bengtmartensson, 2017-04-18:
IMHO, #437 is not "major changes" since it only adds some stuff for not yet supported hardware, and shows a more systematic and clean way of handling hardware dependencies.

More questions:
* Priorities; is clean code preferred over efficiency, or vice versa?
* What API should be offered to the users? How should it be documented?
* Demarcation: i.e., what should _not_ be attempted to solve? 
* Handling of backwards comparability.
* API for protocols; decoding and sending.
* How are new boards to be integrated?
* How should alternative configurations (possibility to select sending pins/timers) be handled? Prefereably without editiing the library code.
* The present "edit the configuration files" is not really "library-ish"
' How is a clean "include only the desired decoders" to be solved, preferrably without editing the library code?

@shirriff did an amazing work in 2009, but the way to honor him is (IMHO :-)) not necessarily to stick to all the decisions he did then. In particular,

* the captured data starts with a useless space (the time from starting until the first mark arrives), and does not end with a space. 
* The parametrization of the protocols is not compatible with the rest of the world.

But, changing this will break backwards compatibility... So possibly having a "classic" version and an "NG" version? 

 I'll stop for now...

closed 2020-10-19 by ArminJo

shirriff, 2020-10-19:
For the record, I don't mind if you change the design :-)