Arduino-IRremote/Arduino-IRremote #390: Help adapting interrupts to STM32 (non-AVR)

TeamDman, 2016-12-06:
**Board:** [Firefly, Generic STM32F103C series](https://github.com/rogerclarkmelbourne/Arduino_STM32)
**Library Version:** 2.2.1
**Protocol:** any

When flashing/verifying the [IRrecvDumpV2 example](https://github.com/z3t0/Arduino-IRremote/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino) the compile fails with the following:

https://gist.github.com/TeamDman/3419b3b5f199ec1a5367bf5fd6ef2a72
```
C:\temp\Dom\arduino-1.6.9-windows\arduino-1.6.9\libraries\Arduino-IRremote\IRremote.cpp:123:5: error: expected constructor, destructor, or type conversion before '(' token

 ISR (TIMER_INTR_NAME)

     ^
```
Relavent: https://github.com/z3t0/Arduino-IRremote/blob/master/IRremote.cpp#L123

Is it possible for me to modify the library to support the board, or is this the only possible interrupt implementation?

z3t0, 2016-12-06:
It should be possible but I would have to look into the board and see what interfaces the creators provide

z3t0, 2016-12-07:
Maybe try replace `ISR` with `isr` probably won't work but just something odd I noticed in the code when doing [this](https://github.com/rogerclarkmelbourne/Arduino_STM32/search?utf8=%E2%9C%93&q=isr)

TeamDman, 2016-12-07:
Still errors with lowercase unfortunately.

I found some timer stuff for the arduino STM32 that looks promising, if I get it working ill be sure to post it here

https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4021c32fdb809f16a61c073d04e1d8c2c487a767/STM32F1/libraries/A_STM32_Examples/examples/Maple/TimerInterrupts/TimerInterrupts.ino

z3t0, 2016-12-07:
It seems like the ISR code is written somewhat differently.

[This](https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/4021c32fdb809f16a61c073d04e1d8c2c487a767/STM32F1/libraries/A_STM32_Examples/examples/Maple/CrudeVGA/CrudeVGA.ino#L59-L63) might be relevant

TeamDman, 2016-12-08:
Through a series of hacks ([commenting most of this out](https://github.com/z3t0/Arduino-IRremote/blob/master/boarddefs.h#L150-L201)) I have managed to make the library 'work' with my board by establishing a timer that calls the interrupt method. However, it seems to hang the program when it attempts to call the digitalRead 

https://github.com/z3t0/Arduino-IRremote/blob/master/IRremote.cpp#L129

Seems like progress, but if the read doesn't work in my interrupt I might be in trouble. I'll see what I can do tomorrow

z3t0, 2016-12-09:
Great, keep us updated then and I will try to pitch in where I can

z3t0, 2016-12-09:
I'll take a look at providing some notes on going about in implementing this over christmas so that it becomes easier on your part

TeamDman, 2016-12-09:
Turns out the schematic file for my board is wrong, pin3 is the only pin that responds to an remote, rather than pin13 as in the schematic

Test I used
https://github.com/TeamDman/Firefly/blob/9b99de5e683573455cbdc2e05a5c3a96954625a5/timertest/timertest.ino

This might be the fix for the hanging issue with digitalread

TeamDman, 2016-12-09:
Got it working for the most part, however it appears that once it receives a code it won't process any others. Example, I push a button on the remote, results.value returns 1599236934 from now on and irrecv.state never changes from 5 (STATE_STOP)

TeamDman, 2016-12-09:
Turns out I forgot to do irrecv.resume() once the signal was printed. Everything seems to be working now! Tomorrow I will work on cleaning up the code so it can actually be used by others

z3t0, 2016-12-09:
Ah excellent thanks for keeping is up to date!

TeamDman, 2016-12-14:
My C knowledge is limited so I couldn't get the header files to behave as a toggle for firefly boards. However, the library can now be included and used for a Firefly board, which was my goal.

Forked repo:
https://github.com/TeamDman/Arduino-IRremote

My changes I made:
https://github.com/z3t0/Arduino-IRremote/compare/master...TeamDman:master
Note, for irRecv.cpp it seems to think I rewrote the entire file?
All I added was a macro for enabling/disabling interrupts at the beginning, and added the timer in the IRrecv::enableIRIn

I hope this can be of help if you decide to officially support these boards

TeamDman, 2016-12-14:
Now that I look at it, it turns out that results.value always returns "4294967295", so I probably messed something up by mistake. Maybe I'm doing it wrong?

LFBFerreira, 2017-03-29:
Hi, i am using a Adafruit Feather M0 Bluefruit, with a Cortex M0 processor, and so far i haven't found a compatible IR library. You seem to be progressing in that way am i correct?

Do you plan to add support for SAMD architectures in the future? Can i do some changes in this library to work for this board in the meantime?

TeamDman, 2017-03-29:
I was not able to make it effectively receive any signals other than "4294967295", however if you only want to trigger an event that should be enough in terms of signal detection. Not sure what to look at for a potential fix

MrBryonMiller, 2017-04-01:
I modified this excellent library for the M0. I started with version 2.2.1 so not up to the latest. But it is possible to use my files with the latest version - just a bit of regression by losing the boarddefs changes.  If people can use this they are welcome.  See this Adafruit forum [https://forums.adafruit.com/viewtopic.php?f=57&t=114520] if your interested. 

Hope I'm not violating anything by doing this in this manner.

z3t0, 2017-04-01:
@MrBryonMiller if you can just paste direct link here I can add the info on the READMe for anyone that might need it

MrBryonMiller, 2017-04-01:
I'm not sure what direct link you want me to paste.  My comment had a link to an Adafruit Forum topic [https://forums.adafruit.com/viewtopic.php?f=57&t=114520](url) (unfortunately it did not come through as a clickable link on my first attempt).  In that topic I included a zip file that had 5 source code files  (irRecv.cpp, IRemote.cpp, IRemote.h, IRemoteInt.h and irSend.cpp).  I explained how a person could download the latest Repository into their Library folder then copy my 5 files on top.  After that they would have a Library that continued to support everything you currently support but also machines with SAMD architectures. 

I would be happy to attach zip file and repeat that information here but I'm not sure if that's what you're asking for.

z3t0, 2017-04-02:
ah sorry I missed the zip file. The link to the forum post is perfect! Thanks I'll add that when possible, or if you want just submit a PR

MrBryonMiller, 2017-04-02:
@z3t0  As you can tell I am brand new to github. I have been reluctant to submit a PR for two reasons; in the event that I totally mangled the PR I did not want to embarrass myself nor did I want to make a lot of work for you.  For instance, my work started with the version just before the boarddefs change.  A compare to the latest version would appear I was backing out boarddefs.  So how do I do a PR and make all this clear? I didn't now.  Yet I realize its work for you to do it so maybe with a little nudge from you I could give it a go.

I also now realize I posted in what is probably the wrong Issue (I followed the comment from @LFBFerreira) A better issue would probably be #274. Should this discussion be over there?

z3t0, 2017-04-02:
For the PR you can just edit a section of the readme to provide a link to your forum post with a sentence description of what it is.

MrBryonMiller, 2017-04-02:
I kinda bumbled along and created a PR.  I could not figure out how to do a Readme change so I added a file explaining what I think you wanted explained.  Please forgive me if I did something wrong.

z3t0, 2017-04-03:
Haha no problem! I'll add it to the wiki once i get some free time. Though that could be a month or two from now as im quite busy at the moment. 

Thank you for the contribution!!

ladyada, 2017-04-09:
@z3t0 @MrBryonMiller hiya folks - barging in here :) i would like to have SAMD support too - z3t0, i can take Byron's code and make a tested PR against current master for SAMD if ya like - (or whatever else would be easiest for you?)

z3t0, 2017-04-09:
@ladyada absolutely! I'm quite inactive at the moment due to studies. But I will actively review and merge PRs.

MrBryonMiller, 2017-04-09:
@ladyada I wanted to mention three things that might make this easier.
1) I learned a little more about github and managed to build a repository with my version of the library. Its at  <a href="https://github.com/MrBryonMiller/Custom-IRRemote">https://github.com/MrBryonMiller/Custom-IRRemote</a>
2) compares against current master are somewhat complicated by PR that spun board definitions into a separate file. Would not have been hard for me to build upon master after boarddefs change but unfortunately I did not have it when I started.  
3) compares against the current master are greatly complicated by the PR that added ESP32 support. That developer and I had to do similar things in defining  the receiver interrupt function and we chose to do it in different places.  Could have been smoother if we built upon each other.

If there was someway to compare against my starting point it would greatly help to isolate my changes. Anything I can do to help, let me know.

ladyada, 2017-04-09:
okie doke! i'll post when i have a ready fork for testing :)

z3t0, 2017-04-10:
@MrBryonMiller following #400 I think the way to move forward with new board definitions for the moment is to add links to them in the readme (through a PR) and I will merge the actual code upstream (with any necessary changes) once I get a chance #386

ladyada, 2017-04-10:
fork is here, tested working with receive (using TC3) and transmit (bitbanging)
https://github.com/adafruit/Arduino-IRremote
if ya can try it out it would be good :)

z3t0, 2017-04-10:
@ladyada once its fairly mature feel free to create a PR like #436

ladyada, 2017-04-10:
@z3t0 how do you feel about a user-exposed interface to allow using alternate timers?

MrBryonMiller, 2017-04-10:
@ladyada  I tested with everything I have which is the following :
-Receive on Uno R3
-Receive and Send on Feather 32u4
-Receive and Send on Feather M0
with the M0 I tried different Send pins and I tried the inverted mode.

Everything compiled and worked as expected except possibly the inverted mode. When inverted is true the output pin is low (which would presumably turn the IR LED on) from power-on until the first Send.  After that the output is as expected - i.e. high while not sending and blipping low during the send.

Probably would benefit from a ESP32 test which I cannot do.  All in all looks pretty dang good to me.

z3t0, 2017-04-11:
@ladyada

I'm definitely doing an entire API overhaul. Essentially I plan o rewriting the library from scratch so I am open to any ideas

ladyada, 2017-04-11:
@MrBryonMiller ok thanks, let me check that out - i may remove inversion for now, since its not terribly common and complicates the pull req

marcmerlin, 2017-04-12:
@ladyada if you need me to test ESP32 when ready, let me know (although you have some too I believe since you tested my ESP32 patch to your neopixel lib)

bengtmartensson, 2017-04-12:
I think I have succeeded in putting all the content in a reasonably clean context, PR #437. Have done some, but not very much testing. (I threw out inversion and selectable sending pin -- this is better handled separately, IMHO).

jordanapplewhite, 2018-04-02:
Hi all! I need support for this lib on Maple Mini (STM32F103). I can hire someone to make it happen if that's possible. Is development on this lib still active? Thanks!

KiLLAAA, 2018-04-08:
@TeamDman, @z3t0 **Hi all**, i made this library partially work on "blue pill" **STM32F103** if someone is still interested. Single button push record value is ok, but for repeated ones only protocol "NEC" is detected, but record value is always FFFFFFFF, might this be a timing issue? see output of IRrecvDump.ino example on serial monitor:
`FF4AB5
Decoded NEC: FF4AB5 (32 bits)
Raw (68): 9150 -4500 600 -550 600 -550 600 -550 600 -550 550 -550 600 -550 600 -550 600 -550 600 -1700 550 -1700 550 -1700 550 -1700 550 -1700 550 -1700 600 -1650 600 -1700 550 -550 600 -1700 550 -550 600 -550 550 -1700 600 -550 600 -1650 550 -600 550 -1700 600 -550 550 -1700 600 -1700 550 -550 600 -1700 550 -550 600 -1700 550 
FFFFFFFF
Decoded NEC: FFFFFFFF (0 bits)
Raw (4): 9150 -2300 550 
`
**is still someone interested to make it work on STM32F103?**

AnalysIR, 2018-04-08:
@KiLLAAA 

The repeat on NEC is essentially empty of data. The original author decided to report this as 0xFFFFFFFF and it has stuck since then as a 'pseudo' standard for NEC repeats.

In summary, your output above is as expected and in line with hte design of the library.

FYI: I suspect the reason for these 'empty' repeats in NEC, was because it save on battery usage, by just sending a brief header/trailer for repeats instead of the full signal, which almost all other IR protocols do.

So looks like you have ot working 100% for NEC on the 'blue pill' :)

jordanapplewhite, 2018-04-08:
@KiLLAAA Hi! I'd be happy to test out your STM32F103 port. Do you have a repo I can clone from? Thanks!

KiLLAAA, 2018-04-09:
@AnalysIR  thats cool! :-D

KiLLAAA, 2018-04-09:
@jordanapplewhite Hi, i forked this repo & added example.. github.com/KiLLAAA/Arduino-IRremote

jordanapplewhite, 2018-04-16:
@KiLLAAA tested and working with an NEC remote control on my Maple Mini clone 👍 👍

TByte007, 2019-07-25:
Any chance the STM32(F103) will be incorporated in the main lib ? The AVR Arduinos are getting really really outdated.

KiLLAAA, 2019-07-28:
@TByte007 it depends... my fork needs some work to fully support stm...
copypaste-> STM32F1 receive only, other STM32 boards untested yet, but it can work too

SidRobo, 2020-05-13:
@KiLLAAA Hi thanks for your works here.
I get this error when i try to compile the example for stmf1. i think this might be because of changes that has happened in newer versions of ArduinoIDE but i have no idea how to solve it.

` \Arduino-IRremote-master\IRremote.cpp:28:10: fatal error: avr/interrupt.h: No such file or directory
   28 | #include <avr/interrupt.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.

exit status 1
Error compiling for board Generic STM32F1 series. `

AnalysIR, 2020-05-13:
As the error suggest its lookng for the interrupt include file for 8 bit AVRs - not STMs.

the offending code is in IRremote.cpp line 28 as the error suggests.

```
#ifndef IR_TIMER_USE_ESP32
#include <avr/interrupt.h>
#endif
```
just comment out these 3 lines temproarily  or add in the logic for your MCU platform using an || on line 28

SidRobo, 2020-05-13:
@AnalysIR hi, unfortunately didn't worked,

[error.txt](https://github.com/z3t0/Arduino-IRremote/files/4624590/error.txt)

AnalysIR, 2020-05-13:
...which may be because your platform is not supported & would require someone with the same platform to make the changes.

The list of supported platforms is on the main page of this library. However, there may may be forks that support it - may be worth checking the links above or searching around.

SidRobo, 2020-05-13:
thanks. I'll do. but this is my board and settings 
[image]

ArminJo, 2020-05-13:
Just try the Arduino Library [IRMP](https://github.com/ukw100/IRMP). It is tested for BluePill!
It is quite easy to migrate :-)

SidRobo, 2020-05-13:
> Just try the Arduino Library [IRMP](https://github.com/ukw100/IRMP). It is tested for BluePill!
> It is quite easy to migrate :-)

thx a lot. ❤

closed 2020-10-17 by ArminJo