Arduino-IRremote/Arduino-IRremote #391: protocol bit length - why is passed as parameter to sendPROT functions?

lucky62, 2016-12-08:
Hello all,

maybe it is a stupid question but why the irsend functions to send IR codes require the nbits parameter?
I think bit length is given by protocol itself - so it should be defined as constant in the protocol definition.

For example:
```cpp
void  IRsend::sendRC5 (unsigned long data,  int nbits)
```
RC5 bit lenght is always 12 bits, so it should be defined as constant parameter and removed from function call - like this:
```cpp
#define RC5_BIT_LENGTH    12
void  IRsend::sendRC5 (unsigned long data)
```
Similarly for other protocols...
Or am I wrong?

Regards
Jan

AnalysIR, 2016-12-08:
You are correct.....

However, that is not to say there are not custom implementations of RC5, out in the wild with other bit lengths, which is probably why it was implemented that way. Regardless, there is probably no record of why it was originally implemented this way.

Similarly for other protocols...the above applies too!

The feature makes things more flexible, particularly if a user wants to create their own protocol version and want to have fewer or more bits transmited than the original protocol.

lucky62, 2016-12-08:
If someone wants to create custom protocol it is better to create new file with new name let say ir_RC5_CUSTOM and put everything here... He can make a copy of existing protocol and make a customization. 
Most probably using only different bit length will not work, because of decoding functions parameters...

By removing the bit length from send functions the code will be smaller, faster, simpler...

AnalysIR, 2016-12-08:
Form one perspective, that would be correct.

However, if we ended up with 1000s of such files, it would be a big(ger) mess.

With a library like this it there are always a trade-off between features, performance, recources etc etc.

Of course, there will be cases where your suggeted approach would be desireable.

It is not my call, just my opinion.

Before putting any time & effort into PRs, you should first check with @z3t0

bengtmartensson, 2016-12-08:
@lucky62:
You are absolutely right; letting the user select protocol length is definitely the wrong thing here. ("If you can't do it right, make it configurable!")

Please have a look at #333; which is more of a background, no need to repeat it here. As you see there, there are more problems than just the payload length.

I believe that Ken Shirriff wrote his stuff in 2009 as a proof of concept. not to be thought as an authoritative guide on how to implement IR protocols. Now we know much more about the IR procols common in customer electronics. There are actually several "competing" libraries out there, [one of those by me](https://github.com/bengtmartensson/Infrared4Arduino). 

Remark. The "Sony protocol" is normally considered three protocols, with payload lengths 12, 15, and 20. So in that context it makes some sense to have the parameter selectable, but still not for all values make sense.

closed 2020-10-17 by ArminJo