Arduino-IRremote/Arduino-IRremote #393: Use aa2408 module

nunomsant, 2016-12-09:
Module: https://anyforsales.com/en/50ma-infrared-emitter-transmitter-module-with-38k-carrier-circuit-for-arduino-aa2804

It's possible tu use a ir module that generates it's own carrier frequency with this library?

In theory we can use any digital output and several emitters in the same board.

What changes are necessary?

Thanks

AnalysIR, 2016-12-09:
You would just have to change the mark function to only turn the IO HIGH, instead of using the PWM to generate the carrier and change the space function to turn the IO LOW.

You would also be restricted to 38kHz. Also, the frequency of that module would be succeptible to drift with temperture. The bonus is that you can use any IO pin for IR Tx.

Another approach would be to look at the cuircuit & short out the carrier generation part. The signal from an Arduino would be much more accurate = better performance & longer range.

FYI: the stated 50mA is a bit of an exaggeration...more like 38mA peak @5V supply.

nunomsant, 2016-12-09:
Ok thank you. This is very helpful, because I need to control several AC systems.

nunomsant, 2016-12-09:
Another question: what range do you think this module can achieve?

AnalysIR, 2016-12-09:
Hard to say...maybe 3-5m, if well pointed. 

You could try putting another 100R resistor in parallel with the one connected to the IR LED to extend the range.

FYI: Most AC units operate on 38kHz.

closed 2020-10-17 by ArminJo