Arduino-IRremote/Arduino-IRremote #394: How to increase RAWLEN in IRremoteInt.h?

Joypacharya, 2016-12-10:
If i decode IR code from AC remote.
At that time displays message.
" IR code too long. Edit IRremoteInt.h and increase RAWLEN".
I tried to increase buffer size upto 255.Still i cant decode my AC code.

How to decode my AC code using your IRrecvDumpV2.

What exactly serial port displayed.
> IR code too long. Edit IRremoteInt.h and increase RAWLEN
Timing[100]: 
     +2900, -1500     + 350, -1100     + 400, -1100     + 350, - 350
     + 400, - 350     + 300, - 450     + 400, -1100     + 350, - 350
     + 350, - 400     + 350, -1100     + 400, -1100     + 350, - 400
     + 400, -1050     + 400, - 350     + 400, - 350     + 400, -1100
     + 400, -1050     + 400, - 350     + 400, -1050     + 400, -1100
     + 400, - 350     + 300, - 450     + 400, -1100     + 350, - 350
     + 350, - 400     + 350, -1100     + 400, - 400     + 350, - 350
     + 400, - 350     + 400, - 350     + 300, - 450     + 400, - 350
     + 300, - 450     + 350, - 350     + 400, - 400     + 350, - 350
     + 350, - 450     + 350, - 350     + 400, - 350     + 350, - 450
     + 350, - 350     + 400, - 350     + 400, - 350     + 300, -1150
     + 350, - 400     + 400, - 350     + 400, -1050     + 400, - 450
     + 350, - 300     + 400, - 350
unsigned int  rawData[100] = {2900,1500, 350,1100, 400,1100, 350,350, 400,350, 300,450, 400,1100, 350,350, 350,400, 350,1100, 400,1100, 350,400, 400,1050, 400,350, 400,350, 400,1100, 400,1050, 400,350, 400,1050, 400,1100, 400,350, 300,450, 400,1100, 350,350, 350,400, 350,1100, 400,400, 350,350, 400,350, 400,350, 300,450, 400,350, 300,450, 350,350, 400,400, 350,350, 350,450, 350,350, 400,350, 350,450, 350,350, 400,350, 400,350, 300,1150, 350,400, 400,350, 400,1050, 400,450, 350,300, 400,350 };  // UNKNOWN 517E0BB5
>

bengtmartensson, 2016-12-10:
IRremote is a very small library for a very small computer. It is unreasonable to expect that it should know "all", or even "most", protocols used by customer electronics.

IrScrutinizer decodes the signal you give as 
protocol = Kaseikyo, device = 2, subdevice = 1, obc = 0, misc = M=35 N=203 G=36

Joypacharya, 2016-12-10:
Is their any other way to resolve this problem?

bengtmartensson, 2016-12-10:
Writing  a renderer and a decoder for the protocol in question, thus extending IRremote?

Not clear exactly what problem you want to solve.

Joypacharya, 2016-12-10:
" IR code too long. Edit IRremoteInt.h and increase RAWLEN". 
How to resolve this problem?

bengtmartensson, 2016-12-10:
Sorry, I misunderstood. I have no idea why rawlen is an uint8_t. Possibly you can try changing it, not sure.

AnalysIR, 2016-12-10:
try the sketch here from our blog....it will be more accurate & you can use the output with IRremote.
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

Alternatively, you can edit the file IRremoteInt.h which is part of the IRremote library and increase the value from 100 to say 230 (max 255). Remember to restart the IDE after the change.

Hnegki7id, 2017-11-27:
**can you all help me? please**

**this is the result of decoding of the remote sharp Air Conditioner.
Button ON**

Raw: (211) 3880, -1816, 540, -412, 544, -1308, 544, -408, 544, -1312, 544, -408, 544, -1308, 544, -408, 548, -1308, 544, -408, 544, -1336, 516, -412, 544, -1308, 544, -1308, 544, -412, 544, -1308, 544, -408, 544, -1308, 544, -1312, 544, -1308, 544, -1308, 548, -408, 544, -408, 544, -1308, 544, -1312, 544, -408, 544, -408, 544, -408, 544, -408, 548, -1308, 544, -408, 544, -408, 548, -408, 544, -1308, 544, -1336, 520, -408, 544, -1308, 544, -408, 544, -408, 548, -1308, 544, -1308, 544, -1336, 520, -432, 520, -436, 516, -408, 544, -1336, 516, -412, 544, -408, 544, -408, 544, -408, 548, -1308, 544, -408, 544, -436, 516, -1312, 544, -1308, 544, -408, 544, -412, 540, -436, 520, -408, 544, -408, 544, -436, 516, -436, 520, -408, 544, -408, 548, -408, 544, -408, 544, -412, 544, -408, 544, -1308, 544, -408, 548, -408, 544, -408, 544, -408, 548, -408, 544, -408, 544, -408, 548, -404, 548, -408, 544, -408, 544, -408, 548, -1308, 544, -408, 544, -408, 548, -404, 548, -408, 544, -408, 548, -404, 548, -408, 544, -408, 548, -404, 548, -404, 548, -1308, 544, -408, 548, -1308, 544, -1308, 544, -1308, 548, -1308, 544, -1308, 544, -408, 548, -408, 544, -408, 544, -1308, 548, -408, 544, -408, 544, -408, 520, 

**after I input into arduino sketch like this:**

#include <IRremote.h>
 
IRsend irsend;

unsigned int  rawData[211] = {-1812, 544, -436, 516, -1336, 520, -408, 544, -1308, 544, -436, 520, -1308, 544, -436, 516, -1308, 544, -412, 544, -1308, 544, -408, 544, -1312, 544, -1308, 544, -408, 544, -1312, 544, -408, 544, -1308, 544, -1308, 548, -1308, 544, -1308, 544, -408, 548, -408, 544, -1308, 544, -1308, 544, -412, 544, -408, 544, -408, 544, -412, 544, -1308, 544, -436, 516, -408, 548, -408, 544, -1308, 544, -1312, 540, -412, 544, -1312, 540, -412, 540, -412, 544, -1312, 540, -1332, 520, -1312, 540, -412, 544, -408, 544, -412, 540, -412, 544, -1332, 520, -408, 544, -412, 540, -412, 544, -1312, 540, -436, 516, -412, 544, -1308, 544, -1308, 544, -412, 540, -412, 544, -408, 544, -412, 540, -408, 548, -408, 544, -412, 540, -412, 540, -412, 544, -412, 540, -436, 516, -412, 544, -408, 544, -1308, 544, -412, 540, -412, 544, -408, 544, -408, 544, -412, 544, -408, 544, -412, 540, -412, 540, -412, 544, -408, 544, -408, 544, -1312, 544, -412, 540, -408, 544, -408, 544, -412, 544, -408, 544, -408, 544, -408, 544, -412, 544, -408, 544, -408, 544, -1312, 544, -408, 544, -1308, 544, -1312, 540, -1312, 544, -1308, 544, -1312, 540, -412, 544, -408, 544, -408, 544, -412, 544, -1308, 544, -408, 544, -408, 492};  // ON

int khz = 38; // 38kHz carrier frequency for the NEC protocol

void setup()
{
  Serial.begin(9600);
}
 
void loop() {
  if (Serial.read() != -1) {
    for (int i = 0; i < 3; i++) {
      
 irsend.sendRaw(rawData,211, 38);
 Serial.println("Sending");
  delay(100); //In this example, the signal will be repeated every 5 seconds, approximately.

}
}
}

**can you guys help me to solve this problem..this is very important ... or do you guys see the error of my sketch so it does not work well?**

AnalysIR, 2017-11-27:
@Hengki-7idbox 

Please note that you have posted this issue twice, which is not good and needlessly wastes time for other people.

It would have been much better if you had just created a new issue for your own problem.

ArminJo, 2020-09-16:
See readme.

closed 2020-09-16 by ArminJo