Arduino-IRremote/Arduino-IRremote #397: PSA: Changing the License to MIT

z3t0, 2016-12-12:
Is anyone opposed to switching the license for all new versions of the library from LGPL to MIT?

henkel, 2016-12-12:
I'm supporting the switch to MIT.

z3t0, 2016-12-13:
Just gonna tag any contributors for their opinion. I apologize in advance for any spam, just want to make sure no one is left it in case they want to voice their opiniont

#119 @fpo @sEpt0r @vk2tds @X-Y @DaAwesomeP @cltnschlosser @crash7 @jan-r @madmalkav @audetto @bessl @lumbric @ram-0000 @safaorhan @hmeine @sstefanov @toddtreece @levsa @bengtmartensson @joshuajnoble @Informatic @ElectricRCAircraftGuy @PaoloP74 @Lauszus @henkel @ivankravets @MCUdude @chaeplin @AnalysIR @shirriff @csBlueChip

bengtmartensson, 2016-12-13:
Let @shirriff decide. Not only is he the original author, but there is still his "smell" over most of the work. If he wanted a copy-left license, that should be respected.

BTW, the way "libraries" in the Arduino IDE works, makes it unciear if there are libraries in the legal sense at all. Recall that an Arduino "library" is not an *.a or *.so or *.dll file, but rather a bunch of files the IDE is *compiling* together with the main program ("sketch"). I may be wrong. But if I am right, that would make [library licenses](https://www.gnu.org/licenses/license-recommendations.html) a bad choice.

Personally, I am a proponent of "copy-left" licenses, with the option of dual licensing for the use in non-free contexts.

csBlueChip, 2016-12-13:
All work I have ever released is MIT licensed. "Free as in free"

If we wish to release as GPL, please understand we MUST conform to the
following little excerpt from the GPL-HowTo on Gnu.org
How to use GNU licenses for your own software

This is a brief explanation of how to place a program under the GNU General
Public License <https://www.gnu.org/licenses/gpl.html>, Lesser General
Public License <https://www.gnu.org/licenses/lgpl.html>, or Affero General
Public License <https://www.gnu.org/licenses/agpl.html>. For the GNU Free
Documentation License <https://www.gnu.org/licenses/fdl.html>, we have
a separate
page <https://www.gnu.org/licenses/fdl-howto.html>.

If you are looking for more detailed information, consider perusing our
list of frequently asked questions about our licenses
<https://www.gnu.org/licenses/gpl-faq.html>.

If you are considering using the GNU Lesser General Public License, please
read the article “Why you shouldn't use the LGPL for your next library
<https://www.gnu.org/licenses/why-not-lgpl.html>” first. The article
explains why it may be better to use the ordinary GPL instead, and how we
would make the decision.

Whichever license you plan to use, the process involves adding two elements
to each source file of your program: a copyright notice (such as “Copyright
1999 Terry Jones”), and a statement of copying permission, saying that the
program is distributed under the terms of the GNU General Public License
(or the Lesser GPL).

The copyright notice should include the year in which you finished
preparing the release (so if you finished it in 1998 but didn't post it
until 1999, use 1998). You should add the proper year for each release; for
example, “Copyright 1998, 1999 Terry Jones” if some versions were finished
in 1998 and some were finished in 1999. If several people helped write the
code, use all their names.

For software with several releases over multiple years, it's okay to use a
range (“2008-2010”) instead of listing individual years (“2008, 2009,
2010”) if and only if every year in the range, inclusive, really is a
“copyrightable” year that would be listed individually; *and* you make an
explicit statement in your documentation about this usage.

Always use the English word “Copyright”; by international convention, this
is used worldwide, even for material in other languages. The copyright
symbol “©” can be included if you wish (and your character set supports
it), but it's not necessary. There is no legal significance to using the
three-character sequence “(C)”, although it does no harm.

You should also include a copy of the license itself somewhere in the
distribution of your program. All programs, whether they are released under
the GPL or LGPL, should include the text version of the GPL
<https://www.gnu.org/licenses/gpl.txt>. In GNU programs the license is
usually in a file called COPYING.

If you are releasing your program under the LGPL, you should also include the
text version of the LGPL <https://www.gnu.org/licenses/lgpl.txt>, usually
in a file called COPYING.LESSER. Please note that, since the LGPL is a set
of additional permissions on top of the GPL, it's important to include both
licenses so users have all the materials they need to understand their
rights.

If you are releasing your program under the GNU AGPL, you only need to
include the text version of the GNU AGPL
<https://www.gnu.org/licenses/agpl.txt>.

If you have copied code from other programs covered by the same license,
copy their copyright notices too. Put all the copyright notices together,
right near the top of each file.

[... 67 lines cut ...]
sample copyright disclaimer; just alter the names and program description
as appropriate:

Yoyodyne, Inc., hereby disclaims all copyright interest in the program
“Gnomovision” (which makes passes at compilers) written by James Hacker.

<signature of Moe Ghoul>, 1 April 1989
Moe Ghoul, President of Vice

We would like to list all free software programs in the Free Software
Directory, including all programs licensed under the GPL (any version).
Please see the Directory web page <http://directory.fsf.org/> for
information and an online submission form.

It is also possible to make your program a GNU package, a part of the GNU
Project. (That's if we like the program—we have to look at it first, and
decide.) If you might be interested in joining up with the GNU Project in
this way, please see our GNU software evaluation page
<https://www.gnu.org/help/evaluation.html> for more information and a short
questionnaire.

But you are welcome to use any of our licenses even if your program is not
a GNU package; indeed, we hope you will. They're available to everyone. If
you'd like to advertise your use of a particular license, feel free to use one
of our logos <https://www.gnu.org/graphics/license-logos.html>.

shirriff, 2016-12-13:
>Let @shirriff decide.

I support whatever licensing decision @z3t0 makes.

>Not only is he the original author, but there is still his "smell" over most of the work.

Uh, is that a good thing or a bad thing?

Lauszus, 2016-12-13:
I'm fine with the decision made by @z3t0 as well.

Informatic, 2016-12-13:
No problem here either. Not that I've done any major contributions... ;)

ElectricRCAircraftGuy, 2016-12-14:
I'm fine with @z3t0's decision too. MIT can be used by a broader context of people.

crash7, 2016-12-14:
Support MIT too, @z3t0  :+1:

PaulStoffregen, 2016-12-24:
I'm good with MIT

rlaferla, 2017-03-07:
+1 MIT

z3t0, 2017-03-08:
Just as an update. I still plan on moving to the new licence and the next major version of this library will be licenced under the MIT.

shirriff, 2017-11-27:
Also, you can remove the "Copyright Ken Shirriff" from the files that still have it.

ArminJo, 2020-10-19:
@z3t0 @bengtmartensson @shirriff @csBlueChip I changed the license with the commit above.
Please veryfy if anything is correct, especially the 3 first lines of the license.txt file. Maybe they can be deleted?
@shirriff I changed all you copyright notes and link to `Initially coded 2009 Ken Shirriff http://www.righto.com`, I hope, this is OK.

closed 2020-10-19 by ArminJo

csBlueChip, 2020-10-21:
Looks good to me

commit 27bb3c11 (2020-10-19): Changed license to MIT Closes #397