Arduino-IRremote/Arduino-IRremote #400: ESP8266F support

tvixen, 2017-01-01:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** RC5

Will it be possible to add the esp8266f to the list of hardware ?
This great little board with 4Mb flash will be great together with your IR library.

Thanks in advance.

PaulStoffregen, 2017-01-02:
Is the question really whether this is possible, or rather who will actually do the work?

tvixen, 2017-01-02:
Well first of all, is it possible ?  
As I know the esp8266f is missing some of the IRremote used interrupts.

tvixen, 2017-01-02:
I've found another project https://github.com/markszabo/IRremoteESP8266  
I'm not sure, but I will have a look at this, as it might have the requested IR library.

nonflammable, 2017-03-05:
[https://github.com/letscontrolit/ESPEasy/tree/mega/src](url)
P035 TX, and P016 RX

marcmerlin, 2017-04-10:
@z3t0 @PaulStoffregen 
It is possible, it works, I've used it, but the current authors are kind of worried about the merge being accepted given that the comparitively simple ESP32 isn't going so well, and ESP8266 is much more invasive.
@z3t0 in this specific case, would you consider listing ESP8266 in your readme and pointing to the other lib, in case it never gets merged?
See also https://github.com/markszabo/IRremoteESP8266/issues/148

z3t0, 2017-04-10:
@markszabo @marcmerlin 

>It is possible, it works, I've used it, but the current authors are kind of worried about the merge being accepted (not counting the work involved) given that the comparatively simple ESP32 isn't going so well, and ESP8266 is much more invasive.

sorry about that! I'm trying to keep a nice clean structure since there are so many different platforms to support.

> @z3t0 in this specific case, would you consider listing ESP8266 in your readme and pointing to the other lib, in case it never gets merged?

submit a PR with a section in the README and i will update it

But more generally I am perfectly happy to adapt implementations for other platforms to the mainstream following the strict (albeit necessary) restrictions we try to set. I am currently busy but I will create a new issue and add this to my todo list.

So essentially @markszabo doesn't need to worry about the "code style". For now just add a note in the README and around May 30 I will go ahead and adapt the code from the other repo into the mainstream. Then I'll release a beta for testing.

Thoughts?

marcmerlin, 2017-04-10:
@z3t0 I totally understand you want to keep the code as clean as possible. Sadly, it makes some ports harder. To be pragmatic, I see 2 options, and both are valid choices
1) we get someone with enough clue to merge difficult architectures while keeping the codebase to your liking. If it's someone other than you, it may be hard but not impossible. 
2) if a port makes things too dirty, it is not accepted and stays as a fork, they can be referenced in your README
Obviously at the end of the day, we're all dealing with the free time of volunteers (including yourself), so if no one can do it sooner, thanks for offering to look at it later.
https://github.com/z3t0/Arduino-IRremote/pull/436 opened at your request.

z3t0, 2017-04-10:
@marcmerlin I think your options are sensible and whichever ends up happening is okay with me.

Also in terms of ports being difficult I plan on writing a guide for writing additions to the library so that things are easier to understand! Unfortunately, again this won't be any time soon

bengtmartensson, 2017-04-17:
I have just committed a fix that I believe is solving the issue, see link above. It requires #437, which is not yet settled. For this reason, no PR yet. But, feel free to test it and to give feedback.  Just check out https://github.com/bengtmartensson/Arduino-IRremote branch esp8266. 

Note: If the project switches to MIT license, there are some issues to clarify here.

marcmerlin, 2017-04-17:
Very cool @bengtmartensson , thank you.
I'm happy to give this a try in a little bit, and I also pointed your work to the folks maintaining the fork, who can probably better comment and test than me.

closed 2020-10-17 by ArminJo