Arduino-IRremote/Arduino-IRremote #401: Add option to cycle through IR codes

GeorgeIoak, 2017-01-04:
If you have a piece of equipment that can be controlled by IR but you don't have the remote it would be nice to have a routine which steps through possible codes. I know that there are a rather large number of possible codes but if you have some information you might be able to narrow down the number of choices.

Just a thought for a possible future feature.

Tag me by typing @z3t0 somewhere in the issue

Tag this issue by typing "#386"

AnalysIR, 2017-01-04:
Just use a for loop, with a 5 second gap between each signal sent & make an appropriate adjustment to the code sent during each iteration of the loop. Printing out the value sent each time would help.

@z3t0 #386

GeorgeIoak, 2017-01-04:
The problem with a simple for loop is the structure of the NEC code format. The 1st byte is the "address code", then ~"address code". The 2nd byte is "command code" and then ~"command code".

I'm far from a good programmer so I was having trouble coming up with a routine that would increment the first byte, and then the complement, and then putting the total code back together again. I just thought if there was a test code case where you could adjust the first byte and the code would reassemble the total code, you then could just use a simple loop.

Hopefully I explained myself well enough.

GeorgeIoak, 2017-01-08:
Here is the code I ended up with which cycles through the possible codes.

`uint32_t lower_part = 0xC03F;
uint32_t codes[0xFF];
bool runXTimes = true;
//byte arr[] = { 0x80, 0x7F, 0xC0, 0x3F };

void setup() {
  Serial.begin(9600);
  Serial.println("Begining the Code Now");
}

void loop() {
  if (runXTimes) {
    delay(15000);
    for (uint32_t value = 0; value < 0xFF; value++) {
      uint32_t compliment = ~value;
      codes[value] = (value << 24) | (compliment << 16) | (lower_part);
      Serial.println(codes[value], HEX);
    }
    runXTimes = false;
  }

  //  unsigned long result = (unsigned long)arr[0]<<24 | (unsigned long)arr[1]<<16  | (unsigned long)arr[2]<<8 | (unsigned long)arr[3];
  //  Serial.print("This is result: ");
  //  Serial.println(result, HEX);

}`

z3t0, 2017-03-26:
I think this might be best implemented as a part of the Sketch, not the library. But I will add an example sketch that explains how this can be done! Thanks

closed 2017-03-26 by z3t0