Arduino-IRremote/Arduino-IRremote #404: Defining pin number

sevano, 2017-01-19:
**Board:** ARDUINO MEGA
**Library Version:** 2.1.0
**Protocol:** RAW

Is it possible to change the pin number in the application. I use IRsendRawDemo. 
example irsignal(3) 
3 = pin

AnalysIR, 2017-01-19:
> Is it possible to change the pin number in the application.

No not during runtime...on AVRs the output PWM pin for IR is fixed based on the Timer being used. You can see the assignments for different MCUs in the README of the library.

You can of course change the timer being used (before compiling) to move the IR output to an alternative Pin.

`ATmega1280, ATmega2560 	5, 6, 9, 11, 46`

On the Mega the default is Pin 9, but you can also select any of the above by changing the Timer being used.

sevano, 2017-01-20:
Thanks AnalysIR

closed 2017-01-20 by sevano