Arduino-IRremote/Arduino-IRremote #407: Commodore Amiga CD-TV Protocol

salocinx, 2017-01-22:
Hi, first of all many thanks for this great library! I am looking for someone that is willing to implement the IR-based "Commodore CDTV Remote Control Protocol". It is a proprietary protocol. I don't have an oscilloscope for analysis and I would not be capable of doing the job even if I would have one. But I could provide a CDTV controller for testing.

I have tested the remote control with this library's the NEC protocol. Button presses are recognized quite well, but there's a problem with the repeat signal, it's somehow not reliably detected by the library (I get the repeat signal only from time to time when holding a button pressed).

Resources I have found so far:

1. [Official Commodore CDTV Manual -> not much Information about the Protocol itself](https://computerarchive.org/files/comp/hardware/amiga/CDTV%20Service%20Manual-ENG.pdf)
2. [A Youtube video series from one that implemented the protocol on a PIC16F628A -> Video 2/5](https://www.youtube.com/watch?v=9wn9slhRh6c)
3. [The source code of the protocol implementation on the PIC16F628A](http://www.picbasic.co.uk/forum/showthread.php?t=21158)

Someone that is willing to implement the CDTV protocol :-) ?

bengtmartensson, 2017-01-22:
There is a [Lirc file](https://sourceforge.net/p/lirc-remotes/code/ci/master/tree/remotes/commodore/CDTV.lircd.conf) for CDTV. It can be imported in IrScrutinizer, revealing that the commands are using the protocol RCA-38(Old), D=8, varying F. IrScrutinizer can also generate [Arduino code directly](https://gist.github.com/bengtmartensson/7f85396339718980d3c6cc4721989eca).

Does this help you?

AnalysIR, 2017-01-22:
@salocinx 
Did you consider just using the RAW format supported by the library? This would be fine if you are just looking to replace the remote. However, if you need to decode signals sent by the remote, then you will need more.

A first step for you to try would be to post the output of a set of signals captured from your remote, using the example sketch **IRrecvDumpV2**, provided with the library.

_PS: It would be unusual to get others to write all the code for you. Typically, users post their own efforts and get assistance/suggestions where needed._

salocinx, 2017-01-23:
@bengtmartensson: Wow, thank you a lot for your efforts. Sure, this will help 👍  Your generated code is for sending the button code as raw data, I will need it the other way around, but it's a great starting point.

@AnalysIR: Yes that's correct. I basically want to receive the CDTV signals, not sending them. My goal is to control a Amiga CD-32 with an Amiga CDTV remote.

I meanwhile started two threads, one in the Arduino forum and the other in the Amiga forum:
http://forum.arduino.cc/index.php?topic=449459.new#new
http://eab.abime.net/showthread.php?p=1136666#post1136666

I also analysed the timings of my CDTV remote with the **IRrecvDumpV2** sketch, here are the results:

`Encoding  : UNKNOWN
Code      : 72A03D6B (32 bits)
Timing[51]: 
     +8900, -4450     + 400, -1200     + 350, - 400     + 400, - 400
     + 400, - 400     + 400, -1200     + 400, - 400     + 350, - 450
     + 350, - 400     + 400, - 400     + 400, - 350     + 450, - 400
     + 350, - 450     + 350, - 400     + 400, -1200     + 400, -1200
     + 350, -1200     + 400, - 450     + 350, -1200     + 400, -1200
     + 350, -1200     + 400, -1200     + 350, -1250     + 350, -1200
     + 400, -1200     + 350
unsigned int  rawData[51] = {8900,4450, 400,1200, 350,400, 400,400, 400,400, 400,1200, 400,400, 350,450, 350,400, 400,400, 400,350, 450,400, 350,450, 350,400, 400,1200, 400,1200, 350,1200, 400,450, 350,1200, 400,1200, 350,1200, 400,1200, 350,1250, 350,1200, 400,1200, 350};  // UNKNOWN 72A03D6B`

Someone from the Amiga forum wrote me some hours ago, that he already has implemented this proprietary Commodore CDTV protocol and managed to add it to the **IRreceive** library. I will now wait for his code before I dig deeper into this issue (although I am looking forward to do so :) ).

Thanks for your PS, I will keep this in mind when posting here in future.

AnalysIR, 2017-01-23:
I have imported you signal into AnalysIR & attached a screenshot.

it looks like a pretty straight forward signal & you could just copy the NEC code from the library and change the NEC timings to match AMIGA (8900/4450/400/1200). Also change the NEC 32 bits to 24 bits. The NEC repeat stuff can be ignored for this signal. As you mentioned it is a 24 bit signal.

[image]

salocinx, 2017-01-23:
Wow great! It now works pretty fine 👍  Thank you! Sometimes I still get some HEX codes I am not sure if they are correct (mainly when pressing multiple buttons together), but I will make some statistics and write all meaningful button combinations down in order to verify.

Why do you think to ignore the repeat signal? Because it should be the same than the NEC one or because I won't need it? I will need the repeat signal to determine if and when the user released a button, this is quite important for a game controller like the CDTV remote.

AnalysIR, 2017-01-23:
Great to hear the progress made. Don't forget to submit you code for inclusion in the library, once you are happy with it.

NEC sends a special/shorter repeat signal, which most other protocols don't send. That is why I suggested leaving the 'special NEC' repeat bit out. Of course if you detect a second signal you should process that as normal.

Of course if the AMIGA has a similar repeat signal to NEC then you should include it.

The contacts on a old remote buttons can often be dirtynoisy and may benefit from a bit of gentle internal cleaning.

salocinx, 2017-01-27:
@AnalysIR : Thanks for your suggestions. I meanwhile managed to implement the CDTV protocol and it now works flawlessly including checksum validation :-) I never made a "pull-request" so far. Could you please advise me quickly how to do that? Which branch should I take take, etc...? Thanks for your generous support!

AnalysIR, 2017-01-31:
@salocinx 

..Sorry I am not familiar enough with the process to advise...perhaps @z3t0 can help

z3t0, 2017-02-04:
@salocinx This [page](https://help.github.com/articles/creating-a-pull-request/) from GitHub's help site should be useful

salocinx, 2017-02-04:
Pull request initiated. Thank you!

z3t0, 2017-03-26:
Will continue in the PR.

closed 2017-03-26 by z3t0