Arduino-IRremote/Arduino-IRremote #409: ATmega2560 - using a timer diferent from Timer2

eduherminio, 2017-01-26:
Hi everyone!

I'm wondering if it's possible to use this library with an Arduino Mega 2560 (ATmega2560) using any timer different from Timer2.
If I'm not wrong, T0 & T2 are 16-bit timers, while T1, T3 & T5 are 8-bit ones; but `delay()` and `millis()` use T0, so the only other 16-bit timer is unfortunately not an option.
Thanks in advance,

Eduardo

AnalysIR, 2017-01-26:
Simple answer is Yes.....please read the README on the home page of the library.

Using the latest version of the library, you need to make a change to the file boarddefs.h (which may be renamed soon). Uncomment the timer you need and comment out all of the other 4 for the mega

Also note that the pin assignments for each timer are different..see README
Finally, using one of the 8 bit timers is fine.

eduherminio, 2017-01-26:
mm right, I misunderstood the README, thanks.

closed 2017-01-26 by eduherminio