Arduino-IRremote/Arduino-IRremote #410: Lost PWM output controll

daimon82, 2017-01-29:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Samsung

Library is working fine, but while it is loaded I can only output PWM signal using pin 5 and 6, the rest of PWM pins is unresponsive.

nm3210, 2017-01-30:
Maybe a duplicate of [251](https://github.com/z3t0/Arduino-IRremote/issues/251)? 

Did you try all the other PWMs? I have had luck with 9 & 10 on the atmega328p (and 5 & 6, as you mentioned). Writing analog/pwm out to 11 totally hosed up the processor though until a restart.

z3t0, 2017-01-30:
Please check point 2 of the following checklist that is a part of the template for any new issue:

```
**checklist:**
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided
- [] The title of the issue is helpful and relevant 
```

daimon82, 2017-01-30:
Sorry for the delay here is the complete [PS 2004 code](https://gist.github.com/daimon82/f20da78598a99d77e4c73e1a273bacf2)

Will check in few hours if I can digitalWrite to the PWM pins at all, but when IRRemote is disabled I can analogWrite to any PWM pin including pin 11 with no problems.

AnalysIR, 2017-01-30:
IRremote uses Timer2 by default on the UNO. This will impact any other use of this timer and the 2 related PWM pins.
If you need those particular pins, you can switch to Timer1 in boarddefs.h.

daimon82, 2017-01-30:
I've changed the boarddefs.h to use Timer1 now working pins are 3, 5, 6
not usable pins are 9 and 10
writing to pin 11 hangs arduino,
Going back to Timer2 makes pins 5, 6, 9 and 10 to work as expected - it was after midnight when i was first testing, so this issue is indeed duplicate of [251](https://github.com/z3t0/Arduino-IRremote/issues/251)

z3t0, 2017-03-26:
The conflict is specific to the use case and as noted has already been covered in #251 

Feel free to reopen if your issue has not been resolved or sufficiently answered.

closed 2017-03-26 by z3t0