Arduino-IRremote/Arduino-IRremote #411: Learning Code Process

brunoudi, 2017-01-30:
I'm using this library to learn some IR codes, but I'm having some difficults to process the raw data.

In my opinion, I should learn the codes 3 times and check if the result are the same. With that, I avoid learn a wrong code, or a noise for example.

The problem, is I get the same correct code 3 times, but with some "insignificant differences". See below:

    unsigned int  rawData[67] = {4400,4600, 550,1700, 550,650, 550,1750, 550,1750, 550,600, 550,1750, 550,650, 550,650, 550,1750, 550,650, 550,1750, 550,1700, 550,600, 550,1750, 550,600, 550,650, 550,650, 550,650, 550,650, 550,650, 550,1750, 550,1750, 550,600, 550,650, 550,1750, 550,1700, 550,1700, 550,1750, 550,600, 550,650, 550,1750, 550,1750, 550};  // SAMSUNG B4B40CF3 

    unsigned int  rawData[67] = {4450,4600, 550,1750, 550,650, 550,1750, 550,1700, 550,650, 550,1750, 550,650, 550,650, 550,1750, 550,600, 550,1750, 550,1750, 550,600, 550,1750, 550,650, 550,650, 550,650, 550,650, 550,650, 500,650, 550,1750, 550,1750, 550,600, 550,650, 550,1750, 550,1750, 550,1700, 550,1750, 550,600, 550,650, 550,1750, 550,1750, 550};  // SAMSUNG B4B40CF3

    unsigned int  rawData[67] = {4450,4550, 550,1700, 550,600, 550,1700, 550,1700, 550,600, 550,1750, 550,600, 550,600, 550,1700, 550,600, 550,1700, 550,1700, 550,600, 550,1750, 550,600, 550,600, 550,650, 550,600, 550,600, 550,650, 550,1700, 600,1700, 550,600, 550,600, 550,1700, 600,1700, 600,1700, 600,1750, 550,600, 550,650, 550,1750, 550,1700, 600};  // SAMSUNG B4B40CF3

I want to learn generic codes, so I pretend to store the raw data, like above.

**The 3 "raw data" are acceptable, but I dont know exactly check that in code.**

For example, I could accept variance up to 50 (or 100), but if all vary, it will be accepted but will not work.

AnalysIR, 2017-01-30:
Don't use the RAW data...let the library do the decoding to HEX & protocol (Samsung??)

See the examples provided with the library.

brunoudi, 2017-01-30:
Considering I will use for different devices, I think the raw data is the best way.. I'm planing to use it it as byte[] ([like this pull request](https://github.com/z3t0/Arduino-IRremote/pull/288))

closed 2017-01-30 by brunoudi

AnalysIR, 2017-01-31:
Tip:  search for **hash codes** with IRremote