Arduino-IRremote/Arduino-IRremote #412: Epson projector support

rlaferla, 2017-01-31:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Epson?

Is there support for Epson projectors in this library?

z3t0, 2017-01-31:
Not specifically... but you could figure out through raw decoding and tools like AnalysIR

bengtmartensson, 2017-01-31:
(We should have a "forum" for this sort of questions)

There are a large number of IR signal resources in the internet. [This](http://harctoolbox.org/IR-resources.html) is my list. Of course, it is also possible to capture the signals yourself.

Using IrScrutinizer, i imported irdb.org's Epson/Projector (it is NEC2 signals) and exported it as "Arduino raw", consisting of an ino file that can be compiled and run directly. [Here](https://gist.github.com/bengtmartensson/e7f475f04d9d8bc596602c46d7654323) it is.

Of course, there are a log of parameters that can be tuned...

AnalysIR, 2017-01-31:
> Is there support for Epson projectors in this library?

Assuming this relates to an IR remote control signal, then it is of course supported by the Library as is, using RAW format. Recognising or decoding it as an Epson singal is another matter.

The decodeHash function could also help in recognition in lieu of full decoding.

@rlaferla once you capture some signals using the RAW method, it shouldn't be too difficult adding full support for it to the library. 

Also, I wouldn't be surprised if your projector is already using a supported format, like 'NEC'....you should try it out and report back.

z3t0, 2017-02-07:
> (We should have a "forum" for this sort of questions)
@bengtmartensson  Do you just mean a support forum for arduino libraries or that sort of thing? I talked to another arduino dev awhile back about something like this and he/she seemed interested and even offered us server resources...

bengtmartensson, 2017-02-07:
@bengtmartensson Do you just mean a support forum for arduino libraries or that sort of thing?

We started a discussion in #387, see also #199 and others. Also, I would welcome some sort of "forum" that is not immediately tied to a particular "product". For example, here often questions are posed, for which I would like to recommend a solution not based on IRremote. In order to respect the nature of the issues; in general I do not reply at all.

rlaferla, 2017-02-13:
Yes, it works using a NEC code.  Thanks everyone!

closed 2017-02-13 by rlaferla