Arduino-IRremote/Arduino-IRremote #416: irremote stops working with raspberry pi 3

venkatyaragalla, 2017-02-10:
I have built pcb with atmega1284p microcontroller with raspberry pi 3 and irremote, which relays(on/off) are controlled by mobile and remote.if i am using only raspberry pi  to control relays,it's working fine. but if i am using  raspberry pi and irremote, I noticed it hangs up after powering on the relay and the leds and not powering down. When they are off it usually does not hang up. But the power drain is not a permanent problem. If I reset the pcb power it works again no matter leds and relay are on or off. 

**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

**Code Block:**
```c

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2017-02-10:
It seems this question is related to electronics rather than IRremote. I think you would be best to post your issue on the Arduino oe EEVBlog forum. You will probably get an answer if you do a search on either forum....might be related to putting a diode across the relay to bypass any spikes when switching the relay.

Your issue does not appear to be related to the IRremote library.
FYI:  @z3t0

z3t0, 2017-03-26:
> Your issue does not appear to be related to the IRremote library.

Agreed.

closed 2017-03-26 by z3t0