Arduino-IRremote/Arduino-IRremote #417: Configure different send pin?

rlaferla, 2017-02-13:
**Board:** Adafruit Trinket 5V
**Library Version:** 2.1.0
**Protocol:** Any

The Trinket uses a Tiny85 and the remote library is configured to use pin 1.  How can I configure it to use a different output pin?  Pin 1 is connected to the builtin LED which is pulsed on startup.

AnalysIR, 2017-02-13:
I am not familiar with the trinket...but did you consider asking the supplier/manufacturer for support on their platform?

rlaferla, 2017-02-14:
Why would the manufacturer need to be involved?  Shouldn't it be easy to remap the pin number?  Just like we use  `pinMode(1, OUTPUT);` we could have `irPinMode(1, SEND);` or something like that.

AnalysIR, 2017-02-14:
They don't need to be involved, but I thought there might be a possibility that they would already have a workable solution available - as they seem to put some effort into IR on at least some of their platforms.

Assuming it is a standard ATtiny85 AVR, you will find the available/supported pins, if you read the README of this library. 

Generally, the target pins are tied to Timers on an AVR. There is usually a **fixed** link between each timer and one or more GPIO pins.

If all else fails do a search for solutions that use a 'bit-banging' approach to generating the IR carrier.

Another approach, is to disable/remove the LED or move it to another pin - thus leaving pin 1 available for IR. Depending on how the existing LED is wired up, you may(?) be able to run the IR off the same pin, by just ignoring the exiting led.

rlaferla, 2017-02-14:
The Tiny85 uses pin 1 (according the README).  For other controllers, the README lists multiple send pins.  How do we choose which send pin?  Or is there only one send pin and it varies by manufacturer?

[image]

AnalysIR, 2017-02-14:
#218 summarises it, in particular the last post.

For ATtiny85, as per the README, only one pin is supported by the library - presumably because the ATtiny85 has only 1 spare Timer with only 1 associated pin.

Did you try contacting the manufacturer/supplier of the module for platform specific support? I just did a quick web search and lots of good stuff was returned, including examples!

rlaferla, 2017-02-14:
Not yet but will do.  Just wanted to clarify everything first.

z3t0, 2017-03-26:
Closing as the issue was not with the library and the reporter either found the solution or another method. Feel free to reopen if that is not the case.

z3t0, 2017-03-26:
Marking this issue as a note to add a section on the README on how to change send pins.

closed 2020-10-12 by ArminJo