Arduino-IRremote/Arduino-IRremote #418: Atmega128

czghzc, 2017-02-15:
**Board:** Atmega128
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

**Code Block:16MHz
#include <IRremote.h>

int RECV_PIN = 9;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  delay(100);
}

Arduino:1.8.1 (Windows 7), 开发板："ATmega128, 2.7v, Disabled (default), 16 MHz external"

D:\Arduino\arduino-1.8.1\arduino-builder -dump-prefs -logger=machine -hardware D:\Arduino\arduino-1.8.1\hardware -hardware C:\Users\CZ\AppData\Local\Arduino15\packages -tools D:\Arduino\arduino-1.8.1\tools-builder -tools D:\Arduino\arduino-1.8.1\hardware\tools\avr -tools C:\Users\CZ\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\arduino-1.8.1\libraries -libraries D:\Arduino\Program\libraries -fqbn=MegaCore:avr:128:BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10801 -build-path C:\Users\CZ\AppData\Local\Temp\arduino_build_434760 -warnings=none -prefs=build.warn_data_percentage=75 -verbose D:\Arduino\Program\ProjectIr\IRrecvDemo\IRrecvDemo.ino
D:\Arduino\arduino-1.8.1\arduino-builder -compile -logger=machine -hardware D:\Arduino\arduino-1.8.1\hardware -hardware C:\Users\CZ\AppData\Local\Arduino15\packages -tools D:\Arduino\arduino-1.8.1\tools-builder -tools D:\Arduino\arduino-1.8.1\hardware\tools\avr -tools C:\Users\CZ\AppData\Local\Arduino15\packages -built-in-libraries D:\Arduino\arduino-1.8.1\libraries -libraries D:\Arduino\Program\libraries -fqbn=MegaCore:avr:128:BOD=2v7,LTO=Os,clock=16MHz_external -ide-version=10801 -build-path C:\Users\CZ\AppData\Local\Temp\arduino_build_434760 -warnings=none -prefs=build.warn_data_percentage=75 -verbose D:\Arduino\Program\ProjectIr\IRrecvDemo\IRrecvDemo.ino
Using board '128' from platform in folder: C:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6
Using core 'MegaCore' from platform in folder: C:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6
Detecting libraries used...
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "C:\Users\CZ\AppData\Local\Temp\arduino_build_434760\sketch\IRrecvDemo.ino.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "C:\Users\CZ\AppData\Local\Temp\arduino_build_434760\sketch\IRrecvDemo.ino.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "D:\Arduino\Program\libraries\IRremote\IRremote.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "D:\Arduino\Program\libraries\IRremote\irPronto.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "D:\Arduino\Program\libraries\IRremote\irRecv.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "D:\Arduino\Program\libraries\IRremote\irSend.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "D:\Arduino\Program\libraries\IRremote\ir_Aiwa.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "D:\Arduino\Program\libraries\IRremote\ir_Denon.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "D:\Arduino\Program\libraries\IRremote\ir_Dish.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "D:\Arduino\Program\libraries\IRremote\ir_JVC.cpp" -o "nul"
"D:\Arduino\arduino-1.8.1\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega128 -DF_CPU=16000000L -DARDUINO=10801 -DARDUINO_AVR_ATmega128 -DARDUINO_ARCH_AVR   "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\cores\MegaCore" "-IC:\Users\CZ\AppData\Local\Arduino15\packages\MegaCore\hardware\avr\1.0.6\variants\standard" "-ID:\Arduino\Program\libraries\IRremote" "D:\Arduino\Program\libraries\IRremote\ir_LG.cpp" -o "nul"
[... cut ...]
                 from D:\Arduino\Program\libraries\IRremote\irRecv.cpp:1:

D:\Arduino\Program\libraries\IRremote\IRremoteInt.h:222:43: error: 'OCIE2A' was not declared in this scope

 #define TIMER_ENABLE_INTR   (TIMSK2 = _BV(OCIE2A))

                                           ^

D:\Arduino\Program\libraries\IRremote\irRecv.cpp:123:2: note: in expansion of macro 'TIMER_ENABLE_INTR'

  TIMER_ENABLE_INTR;

  ^

使用 2.0.1  版本的库 IRremote 在文件夹： D:\Arduino\Program\libraries\IRremote 
exit status 1
为开发板 ATmega128 编译时出错。

AnalysIR, 2017-02-15:
Have you selected the correct mcu/board/platform under tools in Arduino IDE?

r maybe you need to install support for that board in the IDE?

The ATmega128 uses Timer1 in IRremote & the error messages above relate to Timer2, which suggest you might have the wrong board selected.

czghzc, 2017-02-15:
[image]

czghzc, 2017-02-15:
Is there anything wrong with it?

AnalysIR, 2017-02-15:
My guess is that it is related to your board definition in the Arduino IDE.

Unfortunately, I don't have that board to test with.

The obvious issue is that it is trying to access Timer2 instead of Timer1....which suggests that somehow it is not properly defined as an ATmega128...but cannot be 100% certain.

Sorry.

yesyesuk, 2017-02-15:
You don't really need the physical board. You can just select that board in the IDE and then use the Verify button to compile.

czghzc, 2017-02-15:
I found that the "IRremoteInt.h" file did not
Defined (__ AVR_ATmega128__)

// Arduino Mega
#if defined (__ AVR_ATmega1280__) || defined (__ AVR_ATmega2560__)
// # define IR_USE_TIMER1 // tx = pin 11
#define IR_USE_TIMER2 // tx = pin 9
// # define IR_USE_TIMER3 // tx = pin 5
// # define IR_USE_TIMER4 // tx = pin 6
// # define IR_USE_TIMER5 // tx = pin 46

// Teensy 1.0
#elif defined (__ AVR_AT90USB162__)
#define IR_USE_TIMER1 // tx = pin 17
    
// Teensy 2.0
#elif defined (__ AVR_ATmega32U4__)
// # define IR_USE_TIMER1 // tx = pin 14
// # define IR_USE_TIMER3 // tx = pin 9
#define IR_USE_TIMER4_HS // tx = pin 10

// Teensy 3.0 / Teensy 3.1
#elif defined (__ MK20DX128__) || defined (__ MK20DX256__)
#define IR_USE_TIMER_CMT // tx = pin 5

// Teensy-LC
#elif defined (__ MKL26Z64__)
  #define IR_USE_TIMER_TPM1 // tx = pin 16

// Teensy ++ 1.0 & 2.0
#elif defined (__ AVR_AT90USB646__) || defined (__ AVR_AT90USB1286__)
// # define IR_USE_TIMER1 // tx = pin 25
#define IR_USE_TIMER2 // tx = pin 1
// # define IR_USE_TIMER3 // tx = pin 16

// Sanguino
#elif defined (__ AVR_ATmega644P__) || defined (__ AVR_ATmega644__)
// # define IR_USE_TIMER1 // tx = pin 13
#define IR_USE_TIMER2 // tx = pin 14

// Atmega8
#elif defined (__ AVR_ATmega8P__) || defined (__ AVR_ATmega8__)
#define IR_USE_TIMER1 // tx = pin 9

// ATtiny84
#elif defined (__ AVR_ATtiny84__)
  #define IR_USE_TIMER1 // tx = pin 6

// ATtiny85
#elif defined (__ AVR_ATtiny85__)
  #define IR_USE_TIMER_TINY0 // tx = pin 1

// Arduino Duemilanove, Diecimila, LilyPad, Mini, Fio, Nano, etc
#else
// # define IR_USE_TIMER1 // tx = pin 9
#define IR_USE_TIMER2 // tx = pin 3

#endif

Is there a relationship with here?

But I'm trying to add it after still unsuccessful ...

czghzc, 2017-02-15:
Yes, but the compiler is not successful, but if you choose Arduino Nano board can be compiled successfully, but the choice of Atmega128 compiler is not successful

yesyesuk, 2017-02-15:
I meant AnalysIR doesn't need the board ;-)

The "#else" at the bottom covers all other boards that are not specifically defined above. This is where it is specified to use timer2 for your board.

// Arduino Duemilanove, Diecimila, LilyPad, Mini, Fio, Nano, etc
#else
// # define IR_USE_TIMER1 // tx = pin 9
#define IR_USE_TIMER2 // tx = pin 3

AnalysIR, 2017-02-15:
> I found that the "IRremoteInt.h" file did not

@czghzc If you found all of those dfinitiions in IRremoteInt.h then you are not using the latest version of IRremote...please update V2.1.1 via main page.

> I meant AnalysIR doesn't need the board ;-)

I know that, but I did check to see if it was readily available to install the board in my Arduino IDE, but it didnt seem to be readily available (without going to search for the link etc etc), which means I don't have the board or have it configured in my IDE at this point and dont plan to install it because I dont have the MCU variant .

czghzc, 2017-02-15:
MegaCore installation URL
Https://github.com/MCUdude/MegaCore#how-to-install

czghzc, 2017-02-15:
Oh, the success, because I may be through the library manager to install the old library, I re-download the ZIP library, and then manually after the library can be, and thank you for you, and trouble you

AnalysIR, 2017-02-15:
Great...please close issue.

@z3t0 looks like the latest version isn't available via the IDE, yet. I believe I saw this mentioned elsewhere.

closed 2017-02-16 by czghzc