Arduino-IRremote/Arduino-IRremote #419: Send code from unknown encoding?

rlaferla, 2017-02-16:
How do I send a code when the encoding is unknown?  I have a dump of the code/timing below.  What arguments to I pass to sendRaw() if that's the correct function to use?    What value should I use for hz?

`		void  sendRaw     		(const unsigned int buf[],  unsigned int len,  unsigned int hz) ;
`

**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Unknown (if any)

**Code Block:**
```
Encoding  : UNKNOWN
Code      : E318261B (32 bits)
Timing[67]: 
     +9000, -4600     + 500, - 650     + 550, - 600     + 550, - 600
     + 550, - 600     + 550, - 650     + 500, - 650     + 550, - 600
     + 550, - 600     + 550, -1700     + 550, -1650     + 600, -1700
     + 550, -1700     + 550, -1700     + 500, -1750     + 500, -1700
     + 550, -1700     + 550, -1700     + 550, - 600     + 550, -1700
     + 550, - 650     + 500, - 650     + 550, - 600     + 550, -1700
     + 550, - 600     + 550, - 600     + 550, -1700     + 550, - 600
     + 550, -1700     + 550, -1700     + 550, -1700     + 550, - 600
     + 600, -1650     + 550
unsigned int  rawData[67] = {9000,4600, 500,650, 550,600, 550,600, 550,600, 550,650, 500,650, 550,600, 550,600, 550,1700, 550,1650, 600,1700, 550,1700, 550,1700, 500,1750, 500,1700, 550,1700, 550,1700, 550,600, 550,1700, 550,650, 500,650, 550,600, 550,1700, 550,600, 550,600, 550,1700, 550,600, 550,1700, 550,1700, 550,1700, 550,600, 600,1650, 550};  // UNKNOWN E318261B

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

bengtmartensson, 2017-02-16:
There are two ways. Either you send it in raw form (sendRaw) or you have to identify the signal somehow, and hope that it is a protocol that is supported. For the second method, easiest is to paste it into a tool like IrScrutinizer, which will identify it as NEC with device number 0 and function number 69. Unfortunately, IRremote uses its own encoding, as ONE 32-bit number, not three as "the rest of the world", in the opposite bitorder. To cut a long story short, you will get

    sendNEC(0x00FFA25DU, 32);

(first 00 is corresponding to device number 0, next byte (FF) is the complement thereof, following A2 is 69 translated to hexadecimal, and the bits read in the opposite order. Last 5D is the complement of A2)

z3t0, 2017-03-26:
@bengtmartensson sometimes I just wish people would look through the examples before opening an issue!

closed 2017-03-26 by z3t0

vicktr, 2020-06-09:
Hi, I have some problem with this issues. When I press 1 button on remote, I receive HEX code and RAW code. When I press that same button again, the Hex code is the same but Raw code changes. How to fix this problem?

ArminJo, 2020-06-09:
The raw code depends on the internal state of the Arduino timer in relation to the received signal and might therefore be slightly different each time. (resolution problem). The Hex code is the interpreted one which **must be tolerant** to such slight differences!