Arduino-IRremote/Arduino-IRremote #420: iremote in Arduino 101

alanblu, 2017-02-23:
**Board:** ARDUINO 101
**Library Version:** 2.1.0

Hello everyone!!

I want to use this library to create some IR communications between two arduino 101, but when I try to upload the program I find that "avr/interrupt.h: No such file or directory", I want to ask you if there is any development for the Arduino 101 and if there anything I can do to adjust the library so I can use it with this device.

It works with a Intel Curie and with a logic of 3.3v.

Best regards!
 

**Code Block:**
```c

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2017-02-23:
I don't believe this platform is supported as of now. Typically someone with the platform would submit the changes for inclusion in the library. Some board manufacturers like Teensy are proactive in this regard...others not so proactive.

see also #313

z3t0, 2017-02-24:
As @AnalysIR mentioned your board is not yet supported. 

>[] I have searched existing issues to see if there is anything I have missed.

closed 2017-03-26 by z3t0