Arduino-IRremote/Arduino-IRremote #421: The TV reacts to the send command just when I power on or reset the UNO (most of the time)

HassanKanj, 2017-02-24:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Samsung

**Code Block:**
```c

#include <IRremote.h> 

void setup()
{
}

void loop() {
  IRsend irsend;
  irsend.sendSAMSUNG(0xE0E048B7, 32);
  delay(1000);
}

```

I am trying to control a samsung TV using arduino UNO, I was able to retrieve some hex codes successfully from the original remote control, but the problem is that the code will just run once (when I turn on the UNO or on reset) and by 'working' I mean the TV receives the code and changes the channel, then it will stop working (however I noticed that rarely and after a random number of seconds, it may work again, then it stops!) even though the code is inside the loop() function.

these are some observations:

1- I tried to change the delay to different values but with no success
2- it seems that the IR is emitting the code every 1 second (but for some reason the TV isn't reacting to it, except for the 1st time), I tested that by adding an IR receiver to another arduino and I was able to read the signal from the IR emitter using the serial monitor.

any ideas why this is happening and how to fix it?

AnalysIR, 2017-02-24:
Please review the irsend example provided with the library.
[](https://github.com/z3t0/Arduino-IRremote/blob/master/examples/IRsendDemo/IRsendDemo.ino)

You will see some important difference(s) compared to your code.

You should also update to the latest version.
Version - 2.2.1

HassanKanj, 2017-02-24:
I already started with the example code but with no success, I just updated to 2.2.1 and I am now using this code:

```

#include <IRremote.h>

IRsend irsend;

void setup()
{
}

void loop() {
  for (int i = 0; i < 3; i++) {
    irsend.sendSAMSUNG(0xE0E048B7, 32);
    delay(40);
  }
  delay(5000); //5 second delay between each signal burst
}
```

but same thing, the command just run once.. what's weird is that everything is working but only once!

AnalysIR, 2017-02-24:
remove the for statement and the delay(40)..and see what happens

....the example sends the signal 3 times, because SONY requires that. Samsung does not.

HassanKanj, 2017-02-25:
unfortunately, it is still not working consistently, I replaced the Arduino clone with an original Arduino (just to eliminate that the issue is caused by the board), and I removed the for loop and tried to experiment with different delay values, here's a sample of the new code

```
#include <IRremote.h>

IRsend irsend;

void setup()
{
}

void loop() {
  irsend.sendSAMSUNG(0xE0E048B7, 32);
  delay(5000); //5 second delay between each signal burst
}
```

what I noticed is that the TV channel is changing but in an unpredictable pattern.. sometimes there are few seconds between each change, sometimes it is more than 20 seconds and in some cases nothings happen for a while then suddenly I noticed the channel changes, though in all these cases, the running code is the same, the IR emitter is also in the same place.

AnalysIR, 2017-02-25:
Sounds like it could be the emitter circuit...please post the following:

1. A description of your IR emitter circuit
2. A photo of your emitter circuit wired to the Arduino
3. the model # of the IR Led & link to data sheet

Make sure the top of the IR LED is pointing directly towards the TV.

Finally, can you confirm that the TV works OK with the remote control?

HassanKanj, 2017-02-25:
at 1st I bought some IR emitters from an electronics store near my place, I didn't know their exact model, so after reading your last post I tried another IR LED from an old remote (but same issue), then I went and bought some IR LEDs that have a data sheet, and still not working all the time (sometimes it works and other times it doesn't)

so regarding the current circuit:

- Arduino (Genuino) UNO with the IR LED connected to PIN 3 and GND (no resistor since they mentioned it is a 5V LED [though I tried before to add resistor with previous setups but same issue])

- IR LED called JY9415-B5 that's the link: http://www.ekt2.com/products/productdetails/22_INFRARED_EMITTING_DIODE_5MM_B and that's the datasheet link: http://www.ekt2.com/pdf/22_INFRARED_EMITTING_DIODE_5MM_B.pdf

- currently connecting it to a DC adapter (but I tried earlier using USB)

and here's a photo for the circuit:

[image]

Note: the IR receiver is on the bottom right of the TV

and regarding the remote control, yes the TV is working 100% when using the remote control

AnalysIR, 2017-02-25:
The IR LED now seems in order. IR LEDs are not 5V ...more like 1.2-1.5V forward voltage. The 5V refers to the "Reverse Voltage"

However, This is a very bad way to connect an LED to an Arduino & could damage the LED and/or the Arduino.

As a first step, put a 100R resistor in series with the IR LED and try again.
For background see the following post & use pin 3 on your UNO:
[Driving an Infrared Led directly from an Arduino Pin](https://www.analysir.com/blog/2014/10/03/driving-infrared-led-directly-arduino-pin/)

A better circuit would be like the one on this post in our blog: (but use pin 3 on the UNO)
[Example circuit diagram of driving an IR LED with a transistor](https://www.analysir.com/blog/2015/10/12/backdoor-upwm-hack-on-arduino-for-infrared-signals-using-uart/)

HassanKanj, 2017-02-26:
I have tried adding a resistor, but I was still unable to consistently change the TV channel.
however, I solved this issue by modifying the values for the Samsung constants to match the values in this comment: https://github.com/z3t0/Arduino-IRremote/pull/3#issuecomment-13386495

And, it works perfectly!!

I am not sure, if this happened because my samsung TV is 'relatively' old, and the new TVs use the values available in the library, but if not, then the values should be changed in **ir_samsung.cpp**

These are the current (working) values in my **ir_samsung.cpp** file:

```
#define SAMSUNG_BITS          32
#define SAMSUNG_HDR_MARK    4500
#define SAMSUNG_HDR_SPACE   4500
#define SAMSUNG_BIT_MARK     560
#define SAMSUNG_ONE_SPACE   1600
#define SAMSUNG_ZERO_SPACE   600
#define SAMSUNG_RPT_SPACE   2250
```

and thanks @AnalysIR for your help

AnalysIR, 2017-02-28:
Great spot on the timings!....Manufacturers can use several different IR protocols across their range. Just because a protocol is labelled Samsung, doesnt mean it will work for all Samsung devices or TVs.

PS: It looks like you can close the call now.

closed 2017-03-01 by HassanKanj