Arduino-IRremote/Arduino-IRremote #422: Sending using SendNEC, received by other Arduino as LG

MachinesInMotion, 2017-02-26:
**Board:** ARDUINO Pro Mini
**Library Version:** 2.1.0
**Protocol:** NEC

I am trying to replace a crummy remote with this, which reinterprets commands from a much better remote into something an HDMI switch can understand. The one problem is that the HDMI switch can't understand the command. When I run this code with another Arduino watching, it gets the data right, but thinks the command is in the LG specification instead of the NEC one, which is presumably why the HDMI switch isn't responding. Any thoughts?

Thanks in advance, I really appreciate this library.

---
**Code Block:**
```c
#include <IRremote.h>
#include <IRremoteInt.h>
int RECV_PIN = 11;
IRrecv irrecv(RECV_PIN);
decode_results results;
IRsend irsend;
void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop()
{
  if (irrecv.decode(&results))
    {
      Serial.println(results.decode_type);
     Serial.println(results.value, HEX);
     if(results.value == 0x20DF8877)
     {
      delay(100);
      irsend.sendNEC(0x1FE40BF, 28);
     }
    delay(200);
     irrecv.enableIRIn();
     irrecv.resume(); // Receive the next value
    }
}

.....

AnalysIR, 2017-02-26:
try

`irsend.sendNEC(0x1FE40BF, 32);`

NEC is 32 bits...not 28

I also noticed tht your 2 HEX values are different. Is that intended?

If the above soesnt fixt it, please post the output from the IRrecvDumpV2 example

MachinesInMotion, 2017-02-27:
That did the trick. Thank you so much!

closed 2017-02-27 by MachinesInMotion