Arduino-IRremote/Arduino-IRremote #423: Dump and send problem

brunoudi, 2017-02-28:
I'm facing a strange behavior, and I don't know what to do.

I'm trying to get a Air Conditioning code. It seems the dump is OK, but when I try to send I get nothing.

1) I tested my circuit with two TV's of different brands and it is working fine.

2) I increased the raw buffer to 255, and dumped 3 codes from the AC. Here is the data:

```
unsigned int  rawData1[197] = {6200, 7350, 600, 1600, 600, 1650, 600, 1650, 550, 1700, 600, 1650, 600, 1650, 600, 1650, 600, 1650, 600, 600, 600, 600, 600, 600, 600, 600, 600, 600, 600, 600, 600, 600, 600, 600, 600, 1600, 600, 1650, 600, 1650, 550, 1700, 600, 1650, 600, 1650, 600, 1650, 600, 1650, 600, 600, 600, 600, 550, 600, 650, 550, 600, 600, 600, 600, 550, 600, 600, 600, 600, 1650, 600, 600, 600, 1650, 600, 1650, 600, 1650, 600, 1650, 600, 1650, 550, 1700, 550, 600, 600, 1650, 550, 600, 600, 600, 600, 600, 600, 600, 550, 650, 600, 600, 600, 1650, 600, 600, 600, 1650, 600, 1650, 600, 600, 600, 600, 550, 1700, 550, 1700, 550, 600, 550, 1700, 550, 600, 600, 600, 600, 1650, 600, 1650, 600, 600, 550, 600, 550, 1700, 600, 1650, 550, 600, 550, 1700, 550, 1700, 600, 600, 550, 1700, 550, 1700, 600, 600, 550, 600, 550, 1700, 550, 600, 550, 650, 550, 1700, 600, 600, 550, 600, 550, 600, 550, 1700, 550, 600, 550, 1700, 550, 650, 600, 1650, 600, 600, 600, 600, 550, 1700, 550, 600, 550, 1700, 550, 600, 550, 1700, 550, 600, 550, 1700, 550, 1700, 550, 7400, 550}; // UNKNOWN 15D265A1

unsigned int  rawData2[197] = {6150, 7350, 650, 1600, 600, 1650, 600, 1650, 600, 1650, 600, 1650, 600, 1650, 600, 1650, 600, 1600, 650, 550, 600, 550, 600, 550, 550, 600, 650, 550, 600, 550, 600, 550, 600, 550, 650, 1600, 550, 1700, 600, 1650, 600, 1650, 600, 1650, 600, 1650, 600, 1650, 600, 1600, 600, 600, 600, 600, 600, 600, 550, 600, 600, 600, 600, 600, 600, 600, 600, 600, 600, 1650, 550, 600, 600, 1650, 600, 1650, 550, 1700, 550, 1700, 600, 1650, 550, 1700, 600, 600, 600, 1650, 600, 600, 600, 600, 650, 550, 600, 550, 600, 600, 550, 600, 650, 1600, 550, 600, 650, 1600, 650, 1600, 600, 600, 600, 550, 600, 1650, 600, 1650, 600, 550, 600, 1650, 600, 550, 600, 550, 550, 1700, 600, 1650, 600, 550, 550, 600, 650, 1600, 650, 1600, 600, 600, 600, 1600, 650, 1600, 550, 650, 600, 1600, 650, 1600, 600, 550, 650, 550, 600, 1650, 600, 550, 600, 550, 600, 1650, 600, 600, 650, 550, 600, 550, 600, 1650, 600, 550, 600, 1650, 600, 550, 600, 1650, 600, 550, 550, 600, 650, 1600, 650, 550, 600, 1650, 600, 550, 600, 1650, 600, 550, 600, 1650, 600, 1650, 600, 7400, 600}; // UNKNOWN 15D265A1

unsigned int  rawData3[197] = {6150, 7350, 600, 1650, 650, 1650, 600, 1650, 600, 1600, 650, 1600, 650, 1600, 600, 1650, 600, 1650, 650, 550, 600, 550, 650, 550, 600, 550, 600, 550, 600, 550, 550, 600, 650, 550, 600, 1650, 600, 1650, 600, 1600, 650, 1600, 550, 1700, 650, 1600, 600, 1650, 600, 1650, 600, 550, 600, 550, 550, 600, 650, 550, 600, 550, 600, 550, 600, 550, 650, 550, 600, 1600, 650, 550, 600, 1650, 600, 1600, 650, 1600, 650, 1600, 550, 1700, 600, 1650, 600, 550, 550, 1700, 600, 550, 650, 550, 600, 550, 600, 550, 600, 550, 550, 600, 650, 1600, 650, 550, 600, 1650, 600, 1600, 650, 550, 600, 550, 600, 1650, 650, 1650, 650, 550, 650, 1600, 650, 550, 600, 550, 550, 1700, 650, 1600, 600, 550, 650, 550, 650, 1600, 550, 1700, 550, 600, 650, 1600, 650, 1600, 650, 550, 650, 1600, 550, 1700, 650, 550, 600, 550, 600, 1600, 650, 550, 600, 550, 650, 1650, 650, 550, 650, 550, 650, 550, 650, 1600, 650, 550, 650, 1600, 650, 550, 650, 1600, 650, 500, 650, 550, 650, 1600, 650, 550, 650, 1600, 600, 550, 650, 1600, 650, 550, 650, 1600, 650, 1600, 650, 7350, 650}; // UNKNOWN 15D265A1
```

The raw data, for me looks OK. 

I'm sending the code with the function:

```
irsend.sendRaw(rawData1, 197, 38);
```

My AC is Rheem brand, and my controller is like [this](http://g03.a.alicdn.com/kf/HTB1tGptOFXXXXbZXXXXq6xXFXXXf/Portable-Split-Type-Remote-Control-for-Air-Conditional-New-Replacement-Remote-Suitable-for-font-b-Rheem.jpg).

Is there any suggestion to test, or solve this?

AnalysIR, 2017-02-28:
The last signal decodes OK on AnalysIR as Chigo96AC. (Don't be concerned about the different names as many units are rebranded)

So your emitter circuit may be suspect.

Please explain your IR emitter circuit (post a photo/schematic if possible)

Make sure you point the top of the LED toward the AC.

Alternatively, try this sketch which should get you more accurate timings.
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

AnalysIR, 2017-03-01:
TSAL6100 will give you a narrower range. or TSal6200 will be in between the 2. (refer to data sheets)

The IR is related to the current going thru the LED. (not linear)

You could drop the value of those resistors to say 10R or 22R.

You could also put the 2 IR LEDs in series to make the circuit simpler.

See:
[Constant current infrared LED emitter circuit](https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/)

The circuit in this post would be fine, even with 2 LEDs in series
[Basic IR LED circuit](https://www.analysir.com/blog/2015/10/12/backdoor-upwm-hack-on-arduino-for-infrared-signals-using-uart/)

AnalysIR, 2017-03-02:
...I forgot to mention...about hte device you linked. I suspect that is overkill....

Unless you have very demanding requirments (very large room), I would just go with 2 IR LEDs,(1x TSAL6100 and 1xTSAL6200or TSAL6400) and push about 300mA thru them in series. That should suffice for most typical domestic settings. If that doesn't work out, then just add 2 more until it does.

brunoudi, 2017-03-02:
@AnalysIR I got some TSAL6400 for tests, it is much better. I'm waiting TSAL6100 to arrive, but I think it will be the best choice.

What do you mean by hte device?

The original AC remote control works even if I point it in the opposite direction. TSAL6100 could work like that xD

AnalysIR, 2017-03-02:
> What do you mean by hte device?

The device you linked to in your earlier post....with all the IR LEDs

> The original AC remote control works even if I point it in the opposite direction. TSAL6100 could work like that xD

yes, that is primarily a function of the current passing thru the IR LED(s). Try it first with the TSAL6400 you have, as that has the widest angle of the 3.

brunoudi, 2017-03-10:
@AnalysIR thanks for your answers. My emitter is working, but not as I think it should be.

I'm sending signals and comands with a good distance and angle (TV and BluRay for example), except my Air Conditioner.

In the begining I though it was because my Air Conditioner has a 'bad reveicer', but with it's own remote control it works very well. In my opinion, the only explanation for that (I honestly dont know if it is possible) , is because my dumped signal is not 100% precise. Because of that, if I do not point directly straight and near, the AC not get the signal.

Here are two signals I recorded from my AC with your sketch for long signals ([link](https://www.analysir.com/blog/arduino_record_long_airconditioner_infrared_signals_10/)):

```
Raw: (197) 6140, -7352, 592, -1592, 532, -1668, 592, -1612, 592, -1612, 588, -1596, 588, -1612, 592, -1612, 588, -1596, 532, -616, 556, -568, 588, -540, 592, -532, 588, -536, 588, -540, 588, -544, 524, -616, 512, -1668, 560, -1644, 592, -1612, 592, -1592, 532, -1668, 592, -1612, 592, -1616, 556, -1620, 564, -568, 532, -616, 556, -568, 588, -532, 592, -536, 588, -536, 592, -536, 564, -564, 532, -616, 560, -564, 592, -1592, 532, -1668, 592, -1608, 596, -1612, 564, -1616, 588, -1616, 592, -1608, 596, -1592, 532, -612, 560, -568, 584, -540, 588, -536, 588, -540, 588, -536, 564, -1636, 592, -540, 588, -1612, 592, -536, 588, -1616, 588, -536, 592, -528, 592, -1616, 588, -536, 592, -1612, 588, -540, 588, -1616, 556, -568, 588, -1596, 532, -1668, 592, -532, 564, -1636, 596, -1608, 596, -536, 592, -1608, 516, -1668, 588, -536, 536, -1668, 588, -1616, 592, -532, 592, -536, 592, -1608, 592, -536, 588, -536, 592, -1612, 592, -532, 592, -536, 592, -536, 588, -1612, 592, -536, 592, -1608, 596, -532, 588, -1616, 592, -532, 592, -532, 592, -1616, 588, -536, 592, -1608, 592, -540, 588, -1592, 580, -564, 592, -1592, 536, -1664, 592, -7352, 592, 

Raw: (197) 6168, -7328, 616, -1568, 528, -1672, 616, -1588, 616, -1588, 584, -1592, 564, -1636, 624, -1584, 620, -1588, 556, -564, 592, -532, 596, -532, 616, -508, 620, -508, 616, -512, 532, -616, 612, -508, 648, -1536, 536, -1664, 620, -1584, 620, -1584, 620, -1564, 588, -1612, 616, -1588, 616, -1588, 512, -612, 616, -508, 620, -504, 624, -504, 620, -504, 592, -540, 532, -612, 588, -540, 612, -512, 616, -508, 620, -1584, 620, -1584, 512, -1668, 644, -1556, 620, -1588, 640, -1540, 532, -1668, 620, -1584, 644, -484, 616, -508, 592, -536, 536, -588, 608, -540, 616, -508, 648, -1556, 588, -536, 620, -1564, 532, -616, 588, -1588, 564, -588, 512, -616, 612, -1568, 532, -596, 532, -1668, 616, -508, 544, -1656, 620, -512, 616, -1584, 644, -1560, 616, -508, 620, -1584, 612, -1572, 588, -536, 532, -1668, 616, -1588, 620, -508, 616, -1584, 620, -1568, 532, -612, 612, -512, 620, -1584, 556, -568, 612, -516, 616, -1588, 508, -616, 588, -540, 584, -540, 588, -1612, 616, -508, 624, -1584, 560, -564, 588, -1596, 532, -612, 560, -568, 584, -1596, 556, -592, 564, -1616, 536, -616, 532, -1644, 564, -568, 528, -1668, 592, -1612, 592, -7332, 532, 
```

Is it possible, that my signal is not 100% precise and with a distance or angle, it get's 'some noise'?

AnalysIR, 2017-03-10:
@brunoudi 
Both signals you posted decode well with AnalysIR as Chigo96AC signals.
The timings are out a bit but should be OK as is.

So the likely candidates for any issues are: (in order)

1. Operator error
2. IR emitter circuit
3. Firmware being used
4. Less accurate timings

The information you have posted doesn't allow me to be more precise.

bengtmartensson, 2017-03-10:
@brunoudi : Your signals look awfully dirty ("not 100% precise"). For example, there is a plethora of measurements around 600. Errors add up, which explains why it works under "perfect conditions" only. Try cleansed signals.

closed 2020-10-17 by ArminJo