Arduino-IRremote/Arduino-IRremote #424: Other pin Arduino Leonardo?

Eternyt, 2017-03-08:
It is possible to use a pin that is not one of the three in the table in the readme in the section AtMega 32u4? Sorry for the stupid question but I don't understand where is the limitation to the pin. Many thanks!

AnalysIR, 2017-03-08:
The PWM pin assignment for AVRs are tied to the Timers being used. You could always use another Arduino model like the Mega2560, which has more PWM pins & timers available.

However, the IRremote library is limited to those pins mentioned in the README.

Alternatively, Check out the following, articles:
[3 part series showing how to send IR on any pin (AVR based Arduinos running at 16Mhz only)](https://www.analysir.com/blog/2015/05/12/simple-infrared-pwm-on-arduino/) 
You can adjust the timings for other MCUs/speeds with an oscilloscope

[This one lets you use Pin 4 on a UNO, Nano, Micro (but you forfeit the use of Serial, because it uses the UART clock to generate the IR signal)](https://www.analysir.com/blog/2015/10/12/backdoor-upwm-hack-on-arduino-for-infrared-signals-using-uart/)

example usage skteches are provided, but you will have to customise for your own project(s).

Eternyt, 2017-03-09:
Many thanks!