Arduino-IRremote/Arduino-IRremote #429: Use of Library

MaxAtStar, 2017-03-28:
I didn't use the format as this is just a quick question, can I/you provide a download to an edited version of your library for download? I made a tutorial for sending IR codes with an Arduino to easily control multiple LED Strip arrays. I wanted to edit your code so this can be done, and done easily for the end user. Can I provide a download link to your library with my edits if I give credit? Anyways, thanks, I don't know another way to contact you...

kenan-kajkus, 2017-03-28:
Quote drin readme: "Contact

Email: zetoslab@gmail.com Please only email me if it is more appropriate than creating an Issue / [...]"

closed 2017-03-28 by MaxAtStar

z3t0, 2017-04-19:
@MaxAtStar can't remember if I ever helped you out... do you still need anything?

MaxAtStar, 2017-04-20:
You did, thanks