Arduino-IRremote/Arduino-IRremote #431: Fix RobotIRremote conflict

z3t0, 2017-04-01:
Read the last two comments on #98

z3t0, 2017-04-01:
https://github.com/arduino/Arduino/issues/4040

marcmerlin, 2017-04-01:
first, thanks for opening this new issue.
second, very good point about renaming Arduino-IRremote into IRremote on your local disk, it will definitely fix the issue, sadly as you know git clone of this repo will default to the imperfect name.
Maybe a big warning telling people to rename the dir after they git clone and/or some other trick of renaming the include file, or adding a 2nd name for it that can be included instead and get around the silly arduino include ordering issues.
3rd, I owe you a PR and I'm almost done testing it :)

z3t0, 2017-04-01:
If I recall correctly, the way I ended up "dealing" with the issue is by manually renaming the library folder that is in the Releases section. Unfortunately many users come to this website following old instructions from other sites such as Sparkfun and various blogs, many of whom I have contacted. 

The instructions there tell the users to download the repository as a zip, rather then go to releases and download the latest release.

Renaming the repository isnt a good idea imo so that leaves me with not much else todo? Also I really just feel that Arduino needs to nuke their RobotIRremote library that is 3 years old and completely useless

NicksonYap, 2017-04-04:
Thanks @z3t0!
Downloading from Releases solved the issue
https://github.com/z3t0/Arduino-IRremote/releases

marcmerlin, 2017-04-04:
@z3t0 " really just feel that Arduino needs to nuke their RobotIRremote library that is 3 years old and completely useless"
have you talked to them? did you point out that the IR in there is broken and yours is not?
Just wishing that they'd do it, isn't going to make it so :)
Honestly your library is so polished and supports multi-platform that IMO it should be in standard arduino.
If you ask them, maybe they'd be willing to include it instead?

z3t0, 2017-04-05:
@marcmerlin Guess I didn't really consider that. I looked at an issue that's still open about the library conflict and there seemed to be no motion there so I assumed that nothing would come of it. 

I'll add that to my to-do list and contact them sometime soon.

Yansley, 2017-04-08:
Dear IR expert,
I am having the TKD2 issue while compiling. I have removed the folder from the libraries and downloaded the new one and add it to the libraries but still getting:

Infrared_remote_control_car:3: error: 'TKD2' was not declared in this scope
 int RECV_PIN = TKD2;
                ^
exit status 1
'TKD2' was not declared in this scope

marcmerlin, 2017-04-08:
@Yansley you are importing the wrong lib.
Did you rename your downloaded library as IRremote instead of Arduino-IRremote?

z3t0, 2017-04-09:
@Yansley what board

bengtmartensson, 2017-08-22:
In the recently released version 1.8.4, this issue has been resolved: The offending file has been renamed to `RobotIRremote.h`.

Problem solved; close ticket.

closed 2017-08-22 by z3t0