Arduino-IRremote/Arduino-IRremote #434: RC6 sending gobbledygook

Jestopher, 2017-04-06:
**Board:** ARDUINO Nano
**Library Version:** 2.1.0

**Code Block:**
```c

#include <IRremote.h> 

IRsend irsend;
int toggle;
int codeValue;

void setup()
{
  toggle = 1;
  Serial.begin(9600);
}

void loop() {

  for(int i = 1; i<3; i++) {
    irsend.sendRC6(0x0FF28D7, 24);
    delay(40);
  }
  delay(2000);
}

****
I'm trying to transmit RC6 codes from an IR LED on one Arduino Nano to an IR receiver another Arduino Nano.  I'm getting consistent gobbledygook in the serial monitor from the IR receiver when the transmitter is sitting still (and random gobbledygook when the transmitter is in motion).  BUT, sometimes the intended code (FF28D7) IS transmitted/received properly whether the setup is stationary or in motion.  Here is an example readout from the serial monitor:

...
D51C756
D51C756
D51C756
D51C756
D51C756
FF28D7 (Correct!!!)
D51C756
D51C756
D51C756
D51C756
D51C756
D51C756
D51C756
D51C756
9061FABF (?)
D51C756
D51C756
9061FABF (?)
BBE16403 (?)
D51C756
D51C756
D51C756
D51C756
AD8225FC (?)
D51C756
D51C756
C960F370 (?)
D51C756
D51C756
D51C756
...

Nearest thing I can think is that there is some type of timing issue that is **almost** correct, and every once in a while due to subtle variations in the universe, the transmitter spits out codes just within the proper specs to get recognized correctly by the receiver.

I should also point out that I have a real RC6 remote control that I'm using to verify that the receiver end of the setup is working well.  The receiver works beautifully with that.  I can also use the transmitting Arduino Nano to send RC5 codes perfectly.  So it seems specific to RC6.  I have also tried varying the voltage levels to the transmitting LED with the same results at anywhere between 1.4v-1.9v

Help appreciated!

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2017-04-06:
Post the model of IR receiver & link.

How is your IR emitter wired up...driven by Arduino pin or transistor?....post schematic and/or photo.

Other things to try and identify/eliminate the cause...
1. change the baud rate to 115200
2. change the 2 secs to 5 secs
3. remove the for loop and send just one signal at a time
4. make sure the top of the emitter is pointing at/towards the receiver...not too close though

Jestopher, 2017-04-07:
Thanks for the suggestions!  Tried changing baud rate, delay time between loops, and removing the for statement, but still have the same issues.  The emitter is wired to the Arduino pin via a series of resistors to bring the voltage down to ~1.6v
I don't have the model number of the IR receiver - just picked it up at a local mom+pop type electronics store and they just have bins of parts with specs written.  I know if works well though since I can easily send RC5 from the arduino and RC5 and RC6 with existing remotes.

Pics:
[image]
[image]

AnalysIR, 2017-04-07:
some comments
1. the receiver looks like the very cheap Vs1838 variant...which are terrible and would create symptoms like you describe. (I cannot be sure of the model # though)
2. its possible the pinout/wiring for the receiver is wrong from the photo...would need to know the model no for the pinout to be sure. How do you know the wiring for the receiver?...ooops looks like green is not GND so maybe it is OK
3. the total resistance should be 75-100r for the emitter...not clear what values you are using. th led is pointing towards thr ceiling....make sure it points towards thereceiver!
4. how did you measure 1.6v....i would be surprised if you got more than 1.2v driven directly from the gpio pin!

Summary..not much point troubleshooting without proper part numbers...

Jestopher, 2017-04-07:
1. Yeah I have no way of knowing the part number on the receiver or emitter.  The receiver was marked as 38khz, 45degree angle, 2.7-5.5v.  Again, it works fine with the RC6 remote I have, The RC5 remote I have, and the Arduino sending RC5 codes.  
The problem seems to be with sending RC6 via arduino only.

2. Yes the pinout on the receiver is correct; it works perfect with other remotes and protocols.

3.  The emitter works perfectly with RC5 codes.  In use, I have the LED and receiver pointed at each other about a foot apart.  I have roughly 314 ohms in series before the LED (220, 47, 47)

4.  I measured 1.6v by turning the pin to high and adding resistors until I got to 1.6v.  On this board, I am getting 5v from the GPIO pin.  Then I double checked it by setting the code to repeat IR signals repeatedly and testing with a multimeter which showed me that the bursts were topping out at about 1.6v.

Summary - I feel pretty sure it's not a hardware issue.  All parts are working perfectly for sending and receiving except for sending RC6 codes with the Arduino.

bengtmartensson, 2017-04-07:
According to your description, it appears as if the sending Arduino sends flaky RC6 signals, nothing else. I would try sending raw signals (at least for testing). You can generate such Arduino code directly froim IrScrutinizer. I did [this](https://gist.github.com/bengtmartensson/7eedcebd084fcee341a86e75fd54c75a) for you to test.

AnalysIR, 2017-04-07:
> Then I double checked it by setting the code to repeat IR signals repeatedly and testing with a multimeter which showed me that the bursts were topping out at about 1.6v.

The above doesn't make sense for a number of reasons. 
- DMMs typically are not be fast enough to give a reliable reading on an IR signal...but should blip briefly to indicate a signal
- Because the signal is modulated....you would get a fraction of the V drop if it was averaged out _(more like 3->4V because of the lower average drop across the LED)_

However, the test you did using a fixed 5V should be accurate using a DMM.

The 1.6V is also strange (but may be OK for your particular/unkown IR LED).

...for example, on a TSAL6x00 you would be pushing circa 300mA thru it to get 1.6V across the IR LED.

If your measurements are correct....you are getting a max of 11mA thru the IR LED....which is very little.

You should be fine removing the 220R and just using the 2x47R - will get you some extra mA.

**IR receiver:** is 38kHz and RC5/6 are both 36kHz. Shouldn't be a big issue on many 38kHz receivers....but could be more of an issue at very low IR power. Your remotes will be outputting much higher IR current.  The fact that it seems to work with RC5 and not RC6 is also confusing...could indeed be s/w related or just a lucky/unlucky edge condition.

Testing with a RAW signal is a good idea, which will confirm if there is an issue with the library.

closed 2020-10-17 by ArminJo