Arduino-IRremote/Arduino-IRremote #435: IR works in conjunction with fastled on teensy 3.1, and https://github.com/markszabo/IRremoteESP8266 can also do IR at the same time than fastled

marcmerlin, 2017-04-10:
This will not work with the adafruit neopixel lib because it disables interrupts altogether
 
If you use FastLED instead, and that one has more platform specific code. Specifically for teensy 3.1, due, and other ARM (not AVR) 32bit CPUs, it re-enables interrupts in the middle of updating pixels. Those CPUs are fast enough that they are waiting before sending the next signal to the neopixel line, and during that time, there is just enough time for the infrared ISR to run
I have tested with a teensy 3.1, and IR + neopixel, and it works fine.
My code: https://github.com/marcmerlin/Neopixel-IR

I also got it to work on ESP2866, but I'm not exactly sure how/why it works. however, one needs to use this fork of IRRemote:  https://github.com/markszabo/IRremoteESP8266

ESP32 will also work because it can talk to neopixels using the RMT hardware support which works independently from the CPU and therefore does not require interrupts to be ever stopped.
On ESP32, you should use https://github.com/MartyMacGyver/ESP32-Digital-RGB-LED-Drivers for neopixels since they're not supported in fastLED. This works fine at the same time than IRRemote

marcmerlin, 2017-04-17:
Ok, I just spent a bunch of time making a page and video explaining the issues between IR and neopixels, and how to make it work on teensy v3.1, ESP8266, and ESP32:
http://marc.merlins.org/perso/arduino/post_2017-04-03_Arduino-328P-Uno-Teensy3_1-ESP8266-ESP32-IR-and-Neopixels.html
this short video shows how trying this on AVR is an exercise in frustration and handling failures :)
https://www.youtube.com/watch?v=62-nEJtm070

AnalysIR, 2017-04-18:
@z3t0 As that comes up as an issue frequently, might be worth adding a link to readme or similar ?

marcmerlin, 2017-04-18:
@AnalysIR @z3t0 you are more than welcome to take my docs, parts of it, or whatever you see fit and put it in whichever documentation makes the most sense. I did spend all the time writing this up so that others can find it and benefit if they're having the same problems :)

z3t0, 2017-04-18:
If someone could make a pr with the links in a FAQ section of the readme I'll merge.

closed 2020-07-03 by ArminJo