Arduino-IRremote/Arduino-IRremote #438: Is it possible to connect the IR receiver to the hardware interrupt using the library?

mikrob24RUS, 2017-04-13:
**Board:** ARDUINO UNO
**Library Version:** 2.0.1
**Protocol:** ?

**Code Block:**
```c

#include <IRremote.h> 

int IRPin = 12;
IRrecv irrecv(IRPin);
decode_results results;

void setup() {
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    switch (results.value) {
      case 0xFD9867: 
        buzzGo(1);
        break;
.......

    }
    delay(50);
    irrecv.resume();// Receive the next value
  }
/*
Next, a very large software code, which can sometimes last up 
to 2 seconds. If at this time the signal is received from the IR receiver, it is often 
either not readable at all, or is not detected correctly. The project also contains 
a large number of functions and subroutines using timers. Had to change the 
contents of a file IRremoteInt.h, by changing the numbers of the timer.
// Arduino Mega
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
	//#define IR_USE_TIMER1   // tx = pin 11
	#define IR_USE_TIMER2     // tx = pin 9
	//#define IR_USE_TIMER3   // tx = pin 5
	//#define IR_USE_TIMER4   // tx = pin 6
	//#define IR_USE_TIMER5   // tx = pin 46
 With the increase of code project, in the development process, errors and waste 
in the detection of IR codes began to occur more often.
*/

}

```
How to make library to read IR signal not on a timer? Hardware interrupt by connecting the IR receiver to a 2 or 3 pin

z3t0, 2017-04-13:
But how would it keep track of how long a signal is? Or are you asking to keep the timer but use a different receive pin?

mikrob24RUS, 2017-04-13:
Attach the IR receiver to pin 2 or 3, where there is an external interrupt. Create a function attachInterrupt(interrupt, function, mode) which will be calling Your library for signal processing.

z3t0, 2017-04-13:
You should be able to use another pin anyways? Only send pins are fixed as they are tied to timers

AnalysIR, 2017-04-13:
If you need to use interrupts for rx...

See the following post on our blog which provides a sketch for Arduino...
[https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

From memory it uses INT0/INT1, but could be easily changed to use PCINT

mikrob24RUS, 2017-05-15:
Thank you very much for participating in my issue! It turned out that the recognition signal prevent electrical interference affecting the cable going to the IR receiver. Replaced the cable with a shielded and all became good to recognize!
Thanks again!

closed 2017-05-26 by mikrob24RUS