Arduino-IRremote/Arduino-IRremote #439: Interruptdriven signal reception

bengtmartensson, 2017-04-13:
@marcmerlin wrote in #437 (which was unfortunately OT there)

> I was wondering why it was based on a timer that fires no matter what, vs a pin interrupt that would only fire each time the IR signal toggles from low to high or high to low. Is it because on some AVR chips it's just difficult to do this on any pin while on newer chips per pin interrupts are trivial?
> If so, I can see that having 2 versions of the IR receive code, one that is designed to work on pin level change and counting time between each interrupt, vs the one that fires on interval and samples the pin, would probably not be very desirable since it would be 2 different code bases.

(The correct term for "no matter what" is "equidistant sampling".)

IMHO, there is no fundamental problems implementing interrupt driven signal reception. IIRC, it is already done in Chris Young's [IRLib](https://github.com/cyborg5/IRLib). The main drawback is that you introduce new hardware dependencies, a timer and corresponding hardware pin. (OK, the Due is different in that it allows "arbitrary" allocations, are there others?) 

Having said that, the architecture of IRremote makes it very problematic with alternative implementations, have to work with ugly conditional compilation. This is in comparison with an object-oriented approach like in my [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino); you can, at least in principle, add a class as "sibling" to the sampling implementation, cf. [this picture](http://harctoolbox.org/infrared-apidocs/inherits.html). The user can then decide which one of the concrete classes to instantiate.

marcmerlin, 2017-04-13:
Thanks for moving this to another issue. It started with a simple question and got out of hand a bit, sorry about that.
To answer your question ESP32 (and I think also ESP8266) allow pin interrupts on any pin by simply giving a pin number and a callback function, it's super easy.
For instance, see a rotary encoder:
https://github.com/CCHS-Melbourne/iotuz-esp32-hardware/blob/master/Software/Snippets/EncoderTestPinIntrButt/EncoderTestPinIntrButt.ino#L52

On the pin interrupt + timer interrupt, that would only be true if you receive and send. Somehow I'm going to guess that most projects only do one, or the other and therefore would only use a less expensive pin interrupt if they only receive, or would need the timer if they have to send (unless they have RMT-like hardware).
But I'm not sure how much of a difference this would really make on smaller slower chips, while faster chips can't really tell the difference, except for freeing up a hardware timer if you're only receiving.

z3t0, 2017-04-13:
@marcmerlin sorry if I'm misunderstanding something but how would the pin interrupts keep track of signal length without using a timer?

marcmerlin, 2017-04-13:
@z3t0 you'd store the nanos() counter value captured at each pin change into a fifo array with pointer to the last modified added value in the array, modulo array size.
Then when you ask for "was an IR code received" from user code, it would look at that array and see if indeed there is a fully decoded value in there.
Note that I'm not saying "it's easy, why don't you just do that?" :) but more "would it work and be more efficient to do this?"
I think it would work, I think it would be more efficient in reducing interrupts and freeing a timer if you do IR receive only, but it this is all supposition, and again it would be a non trivial code change.

z3t0, 2017-04-13:
Okay, I will add this to my todo list. One thing to look into is the accuracy of nanos() and whether the call itself is resource intensive.

#386

bengtmartensson, 2017-04-13:
> a timer and corresponding hardware pin. (OK, the Due is different in that it allows "arbitrary" allocations, are there others?)

To answer myself, in the language of the documentation for [attachInterrupt](https://www.arduino.cc/en/Reference/AttachInterrupt), it is the question on if `attachInterrupt` allows "arbitrary" pins as first argument; which, again according to that page, is the case for Due, Zero, MKR1000, and 101. @marcmerlin says that it is the case for ESP32 and ESP8266 too. Probably that goes for most of the "new & hot" boards too. 

> only use a less expensive pin interrupt ... 

I think the biggest gain would be a resolution better than the present 50 micros. In particular if you are capturing an "unknown" signal with the purpose of analyzing it, not just recognizing one signal from an a priori known protocol. 

micros() should be more than enough, at least if you are using a demodulating receiver (TSOP****).

AnalysIR, 2017-04-13:
> micros() should be more than enough,

I agree, using the micros() equivalent on the ESP should be good enough for time measurements and more importantly - consistent with the library. 

The 50uSec resolution for sampling is I believe already configurable (`#define USECPERTICK 50`) and could be defaulted to faster for ESP32 & other MCUs if thought desireable. (or just highlighted as an option for users of faster/newer platforms)

FYI: micros() on standard Arduino is @ 4uSecs granularity which is more than good enough for use with standard IR Receivers.

marcmerlin, 2017-04-13:
Sorry I wrote nanos() instead of millis() when indeed micros() was what I should have written.

ArminJo, 2020-10-17:
[IRMP + IRLremote + IRLib2](https://github.com/ukw100/IRMP#quick-comparison-of-4-arduino-ir-receiving-libraries) support interrupt driven decoding, but this is not possible for all protocols.

closed 2020-10-17 by ArminJo