Arduino-IRremote/Arduino-IRremote #444: Reorganizaton of boarddefs.h

bengtmartensson, 2017-04-17:
Sort-of continuation of #437.

It turns out that the stuff that must be added for supporting a new board is sprinkled all through the file. Probably this is due to the time when all processors where AVR, and most configurations were dependent on the timer selected. 

To address this problem, I would like to suggest a reorganization like:

    // Default definitions that only a few boards like to change

    #if defined(BOARD1)
    // stuff for BOARD1
     #elif defined(BOARD2)
    // ...
    ....
    #else
    #error No board recognized
    #endif 

    // Fallback definitions, like
    #if defined(SEND_PIN) & !defined(SENDLED_ON)
    #define SENDLED_ON(pin) digitalWrite(SEND_PIN, HIGH)
    #endif 

alternative structure: Everything that concerns a certain board is in a certain file, i.e.,

     #if defined(BOARD1)
     #include "board1.h"
     #elif defined(BOARD2)
     #include "board2.h"
    ....
    #else
    #error No board recognized
    #endif 

    ....

 Feedback wanted.

z3t0, 2017-04-17:
I was thinking along the same lines and completely agree. It's like having "cores" for the different boards. 

Does arduino support libraries having an internal folder structure because I can see the top level folder getting even messie than it currently is. Though this isn't a real issue..

I support the idea.

bengtmartensson, 2017-04-17:
> Does arduino support libraries having an internal folder structure because I can see the top level folder 

AFAIK, the Arduino environment does not support include paths in the sense of the -I option to the compiler. However, you can use `#include "subdirectory/file.h`. This _is_ portable, i.e. works also on those "backwards OS using a backward slash" :-).

bengtmartensson, 2017-08-17:
Additional requirement:

* If no suitable pre-processor symbols identifying the board is found, an error should be thrown. Presently, instead  a "Uno-like" board is assumed.

Using defaults in situations like this is simply very bad programming style. It is likely to cause problems.

Yes, this surely break some ancient stuff...

bengtmartensson, 2017-08-23:
I started to work on this, but came to the conclusion that it is not a good idea after all. The reason is that the file contains so much "cruft" (in particular ancient boards, the support is written by people (probably) no longer active) that there is no possibility to test a re-organized version. Instead, I propose "just" to clean up and document (in the sense of Doxygen, #461) the file. The remark immediately preceding this still makes sense, though.

closed 2020-06-25 by bengtmartensson