Arduino-IRremote/Arduino-IRremote #445: Formatting guidelines requested

bengtmartensson, 2017-04-17:
There are currently no code formatting guidelines in e.g. Contributing.md. The original code is, though consequently, formatted in a pretty original way ("Shirriff style"). Later additions seem to use the formatting that the current author liked.

For the record, my code in #437 if formatted using the "Linux" scheme (which, I would like to point out, is NOT my favorite), except for CPP commands, which follow the "Shirriff" style.

z3t0, 2017-04-17:
If everyone is okay I think we should choose a standard that is used commonly and has checkers available. That way we can enforce the guidelines without fixing them by hand

z3t0, 2017-04-17:
https://github.com/cpplint/cpplint this follows Google's C++ style format. also can be configured for our own preferences.

bengtmartensson, 2020-06-25:
Has been implemented in current Contributing.md, therefore closing.

closed 2020-06-25 by bengtmartensson