Arduino-IRremote/Arduino-IRremote #448: RCA TV will respond to android app, but not same codes from universal remote.

YoItsTrev, 2017-04-19:
**Board:** Arduino Mega2560
**Library Version:** 2.2.3
**Protocol:** RCA

I am at a complete loss. In 2010 I bought a RCA L40FHD41 40" HDTV. It came with this remote: http://www.remotecontrols.com/product/276045.html

Right after I bought it, I tried to get it to work with our DirectTV remote ( https://www.xfinity.com/support/remotes/  the one with the grey OK/Select button ), but it would not work. Did tons of research, and all I found was people saying they had gotten in touch with RCA and asked them about it, and they were told that the TV used new technology, and that current universal remotes would not work, but future ones would, once they integrated the technology. So, I am at a loss... 

I had my TV in storage, and when I got it back, the remote was missing. I tried using a new Xfinity universal remote ( https://www.xfinity.com/support/remotes/ the one with the red OK/Select) button, but again, nothing worked. My brother in law suggested I used an android universal remote, but my phone nor my tablet has an IR Blaster, so I asked my friend. who I am staying with atm, if his phone does. His does, and he was able to find a IR remote app that worked with my TV. So I fired up the example sketch found here: https://arduino-info.wikispaces.com/IR-RemoteControl and I wrote down all the IR codes, that the app broadcast, after verifying that they all worked on my TV.

Once I did that, just for curiosity sake, I went and keyed in the most often used RCA code 11447 into the universal remote, and look at what the keys decoded at, and the vast majority matched what my friends phone output, but the universal remote WILL NOT control my TV. So, I am at a loss as to what is up. Is the RCA using a higher or lower frequency than standard IR remotes? If so, is it possible to broadcast like that using an Arduino?

z3t0, 2017-04-19:
Just to be clear are you trying to find a protocol that matches your TV? Or do you have access to an existing remote or other that you can use to find the correct code / frequency needed for your TV?

The former may be a lot more trial and error than the latter

YoItsTrev, 2017-04-19:
Half and half. I currently have access to a phone with an IR-Blaster that has an app that will correctly control my TV. But I will not have access to it for long. I will be leaving in less than a week. I wanted to note down the codes that his phone was outputing, and come up with an arduino based solution to replace the remote, since I already had the hardware on hand. I also need to try other codes that the universal remote has, because if you look at the image of the remote here: http://www.remotecontrols.com/product/276045.html

There is a "Format" button that controls the TVs aspect ratios, like wide screen, SD, full, stretched, etc. The app on his phone has no such button, but the many other buttons on the universal remote might allow me to figure out the code to change it. In my original post, the first IRrecvDumpV2 output was for the universal remote. That DOES NOT work on my TV. It will not turn it off or on. The second output, the power button from the android app, DOES turn my TV on and off. What is different between the 2 signals?

z3t0, 2017-04-19:
> His does, and he was able to find a IR remote app that worked with my TV.

That would be one entry point to try and solve this. Can you provide more information about which app, what settings if any? Screenshots would help here as I do not have an android device.

YoItsTrev, 2017-04-19:
Sadly, no. He has the phone in his room, and he is asleep for the night. All I was able to get was the hex codes that it puts out for the various buttons, as well as the IRrecvDumpV2 output for several of the buttons from the app, and I can provide the univeral remote IRrecvDumpV2 outputs for comparison if necessary.

z3t0, 2017-04-19:
Okay hmm.

It would be a good guess to assume that the app is using LIRC

http://lirc.sourceforge.net/remotes/rca/

So I guess lets fish through that directory to see if we can find the correct spec

bengtmartensson, 2017-04-19:
RCA often use a 58kHz carrier, which may explain the problems. 

Anyhow, ignoring the frequency for a while (remember, IRremote cannot measure it (in particularly not with a demodulating sensor :-)), IrScrutinizer/DecodeIR decodes the signal (both actually) you have given as protocol = RCA, device = 15, obc = 42. I think this pretty common for RCA TVs. Codesets extending this signal are widely found on the internet. For example irdb.tk. Using their codes, I geneated Arduino code using IrScrutinizer, found [here](https://gist.github.com/bengtmartensson/71a675942d645f87d6c10c455a862cc2#file-gistfile1-txt) Works? (If not, you can as experiment try changing 58000 to e.g. 38000.)

YoItsTrev, 2017-04-19:
I see a lot of sendRaw in the code. I just woke up, so my brain is at about 30%, but I assume it is to send the signals to the TV, so I will need to hook up a IR LED, but I cannot seem to find a pin definition in the code. What pin do I need to hook the LED to? Or what line do I need to change to change it?

YoItsTrev, 2017-04-19:
I did ask my friend about the app he is using, and he said it wasn't some app he downloaded from the Play Store, it came on his LG phone.

closed 2020-10-17 by ArminJo