Arduino-IRremote/Arduino-IRremote #449: Greater integration with IrScrutinizer

z3t0, 2017-04-19:
It seems that many common issues are solved/debugged using IrScrutinizer.

Might be a good idea to both support that and link to it on the README as a useful tool. 

Eg. #448

bengtmartensson, 2017-04-23:
There are two things here, integration and docu/links.

Integration: I just PRed (#460)  an "example" that allows IRremote to be used directly with IrScrutinizer. Anything else you have in mind? Possibly have IRRecvDemo to produce a more sensible output (space between the sign and the following number...)?

Docu: I am of the standpoint that the present project is _one_ project in the Free Software universe. It is not the universal solution to all problems ("If all you have is a hammer, everything looks like a nail.") Mentioning other free software that does useful things is here natural. Personally, I have no problems with, in a suitable context, also to mention (link to) non-free software. I do not like to "endorse" or recommend some stuff and not other stuff (as we had in a discussion on hardware).

bengtmartensson, 2020-10-17:
Three and a half years later:

#460 has been merged. 

> Possibly have IRRecvDemo to produce a more sensible output

IRreceiveDumpV2 now produces Pronto Hex output, which can be directly cut-n-pasted into IrScrutinizer.

OK to close, IMHO.

closed 2020-10-17 by ArminJo