Arduino-IRremote/Arduino-IRremote #450: MKR1000 IRremote can't send

beaglesparks, 2017-04-20:
**Board:** MKR1000
**Library Version:** 2.2.1  IRremote
**Protocol:** Samsung

I’m new at this so sorry if I’m asking stupid stuff but I need some help.

I’m using the IRremote library with an MKR1000 and I can get  receive and decode to work.  My problem is with IRSEND.  By adding debug statements I’ve determined when I call it, IRsend::enableIROut freezes at line  

 TIMER_CONFIG_KHZ(khz);

I modified IRremoteInfo.ino by adding this

#ifdef IR_USE_TIMER_MKR
     Serial.print(F("Timer defined for use: ")); Serial.println(F("Timer_MKR")); flag = true;
#endif

and got the below.     Suggestions?

BTW I took the liberty of correcting some minor spelling errors in dumpFooter at the bottom of this email. I haven't figured out how to formally suggest changing the code and don't want to screw things up.

Thanks in advance
Phil Kramer

IRremoteInfo - by AnalysIR (http://www.AnalysIR.com/)
             - A helper sketch to assist in troubleshooting issues with the library by reviewing the settings within the IRremote library
             - Prints out the important settings within the library, which can be configured to suit the many supported platforms
             - When seeking on-line support, please post or upload the output of this sketch, where appropriate

IRremote Library Settings
=========================
RAWBUF: 101
Timer defined for use: Timer_MKR
IR Tx Pin: 12
MCU Clock: 48000000
MCU Platform: MKR1000
Mark Excess: 100 uSecs
Microseconds per tick: 50 uSecs
Measurement tolerance: 25%
Minimum Gap between IR Signals: 5000 uSecs
Arduino IDE version: 1.8.1
Debug Mode: ON

IR PROTOCOLS  SEND     DECODE
============= ======== ========
RC5:          Enabled  Enabled
RC6:          Enabled  Enabled
NEC:          Enabled  Enabled
SONY:         Enabled  Enabled
PANASONIC:    Enabled  Enabled
JVC:          Enabled  Enabled
SAMSUNG:      Enabled  Enabled
WHYNTER:      Enabled  Enabled
AIWA_RC_T501: Enabled  Enabled
LG:           Enabled  Enabled
SANYO:        Disabled Enabled
MITSUBISHI:   Disabled Enabled
DISH:         Enabled  Disabled
SHARP:        Enabled  Disabled
DENON:        Enabled  Enabled
PRONTO:       Enabled  (Not Applicable)

Notes: 
     - Most of the seetings above can be configured in the following files included as part of the library
     - IRremteInt.h
     - IRremote.h
     - You can save SRAM by disabling the Decode or Send features for any protocol (Near the top of IRremoteInt.h)
     - Some Timer conflicts, with other libraries, can be easily resolved by configuring a differnt Timer for your platform

void dumpFooter() {
  Serial.println();
  Serial.println(F("Notes: "));
  Serial.println(F("     - Most of the settings above can be configured in the following files included as part of the library"));
  Serial.println(F("     - IRremoteInt.h"));
  Serial.println(F("     - IRremote.h"));
  Serial.println(F("     - You can save SRAM by disabling the Decode or Send features for any protocol (Near the top of IRremoteInt.h)"));
  Serial.println(F("     - Some Timer conflicts, with other libraries, can be easily resolved by configuring a different Timer for your platform"));
}

z3t0, 2017-04-21:
> and got the below. Suggestions?

What was the output to your #if def modification? (I think it got cut out in your post above?)

beaglesparks, 2017-04-21:
it's near the top.  it was

Timer defined for use: Timer_MKR

z3t0, 2017-04-21:
Have you tried the latest release (not the master branch)

beaglesparks, 2017-04-21:
again i'm new at this so i'm not sure.  i'm using 2.2.1 at 

https://github.com/agn0stico/Arduino-IRremote-MKR1000

z3t0, 2017-04-21:
Ah I see, two things to keep in mind.

That is an unofficial port which only supports [receiving](https://github.com/agn0stico/Arduino-IRremote-MKR1000/commit/14523d8672e2d96db0650c7f75bd1e994b2b2f1e) it has not yet been merged upstream into the official project because it breaks the standard we try to follow in this library to minimise the performance hit. i.e.. it makes calls to `digitalWrite()` and because it has not yet been tested thoroughly.

beaglesparks, 2017-04-21:
so do you have a suggestion for me to try to send using the MKR1000?

z3t0, 2017-04-21:
[Arduino-IRremote-master.zip](https://github.com/z3t0/Arduino-IRremote/files/948319/Arduino-IRremote-master.zip)
See if this works, not sure though.

beaglesparks, 2017-04-21:
when i use that i get 

Arduino: 1.8.1 (Windows Store 1.8.1.0) (Windows 10), Board: "Arduino/Genuino MKR1000"

J:\Users1\Phil\Documents\Arduino\libraries\Arduino-IRremote-master-2-2-3\IRremote.cpp:129:5: error: expected constructor, destructor, or type conversion before '(' token

 ISR (TIMER_INTR_NAME)

     ^

z3t0, 2017-04-21:
Ah, then there isnt support for mkr1000 on that fork hmm

z3t0, 2017-04-21:
Unfortunately that just means no one has added a port of this library for that board :(

beaglesparks, 2017-04-21:
right.  it seems like the MKR1000 fork works for receive but not send. IRSend (Samsung) IRsend::enableIROut freezes at line

TIMER_CONFIG_KHZ(khz);

beaglesparks, 2017-04-21:
should i be writing the person who wrote the MKR1000 fork or is that you?

z3t0, 2017-04-21:
@beaglesparks you can ask the other person as they were the one that implemented but two things to keep in mind:
- contributing to the project is completely voluntary
- the fork doesnt support sending yet.

bengtmartensson, 2017-04-22:
Hmm, I have not been reading that carefully, but it sees to me that the MKR1000 is a SAMD architechture. and you should be able to use the fork of #437. That includes sending support through generation of carrier in software ("soft carrier").

beaglesparks, 2017-04-22:
@bengtmartensson - i think that's the one i'm using Arduino-IRremote-MKR1000  2.2.1 at

https://github.com/agn0stico/Arduino-IRremote-MKR1000

again i'm new at this.  if i'm navigating github wrong then maybe i don't have the right one

z3t0, 2017-04-22:
@bengtmartensson the zip i shared above is your fork #437

z3t0, 2017-04-25:
Sorry we couldn't help solve your issue!

beaglesparks, 2017-04-25:
hoping in the future you'll find time to support MKR1000.   good day

andrea689, 2019-12-05:
this library works with MKR1000:
https://github.com/cyborg5/IRLib2