Arduino-IRremote/Arduino-IRremote #455: write a parser for keywords.txt

z3t0, 2017-04-23:
far too annoying to have to change this file for every update. need a consistent api and autogenerator for this

bengtmartensson, 2017-04-23:
Alternative 1: Who cares? Don't waste your time.

Alternative 2: Use Doxygen (cf. #461) to generate XML "documentation". Then write an XSLT-transformatioin that takes the XML documentation to keywords.txt. Should really not  take too long for someone who knows XSLT (there is at least one person around....)

z3t0, 2017-04-23:
Perfect solution.

bengtmartensson, 2017-08-21:
[Here](https://github.com/bengtmartensson/KeywordsTxtGenerator) is such a generator, constructed along the lines in "Alternative 2".

ArminJo, 2020-10-12:
Nor really required, the keywords.txt is changing slowly (and who cares if its content is outdated?)

closed 2020-10-12 by ArminJo