Arduino-IRremote/Arduino-IRremote #456: Proposed rename of the library

z3t0, 2017-04-23:
@shirriff @AnalysIR @bengtmartensson @marcmerlin 

Hi all I am thinking of renaming this library to something like InfraredRemote or anything very similar. The intention being to change the file IRremote.h into something else... far too many distributions of arduino (including platformio) ship with RobotIRremote.

Thoughts?

AnalysIR, 2017-04-23:
I think there is a **lot** of value in retaining the existing name in terms of:
- Search engines
- Use in existing projects, their descriptions and other online content
- References & links in existing discussion groups/Forum etc
- 'Brand'
- The clever thing about the name is that 'it does what it says on the tin' _(reference to a TV ad campaign)_ ,,, as does your suggestion.
- & I guess history.....

Having said that, your point is also valid. 

Are there any similar examples (with solutions) for other libraries?

How is it handled with forks of IRremote? (like Photon or ESP etc)?

What will happen when someone does the same with any new name.?

Unfortunately, I am not up to date with the requirements for the Arduino library management stuff etc to offer any better suggestion. Presumably having something like IRremoteV3 or IRremote3 or IRremote_Official or Original_IRremote etc  would be an issue also.??

So like many things it may be a trade-off. My _(not so strong)_ preference would be to stick with using 'IRremote' as part of any new name unless the issue you mentioned cannot be resolved by other means. 

**PS:** _I really don't understand why the creators of robotirremote, don't resolve the issue....'community spirit' etc. etc._

z3t0, 2017-04-23:
Ideally I am still pushing for removing robotirremote. I have created an issue on platformio and hopefully I can get a response there. The arduino team hasn't gotten back to so I may just try and contact them by email.

bengtmartensson, 2017-04-23:
I think AnalysIR has mentioned most of the arguments. Let me add that, as I foresee, it might happen that "we" want to make incompatible changes and therefore indicate this by another name (or version). Like Python3... That might mean pushing yet another name onto the community again...

If another name, how about -- inspired by the twitter name of **** -- realIRremote :-).

marcmerlin, 2017-04-23:
Didn't we establish that keeping this called IRRemote would be fine enough as long as the directory you get from git is also called IRRemote and not Arduino-IRRemote? 
If the former is the case, at least the arduino IDE knows to get IRRemote.h from the right library.
But yes, the robotirremote stuff should also die :)

z3t0, 2017-04-23:
> Didn't we establish that keeping this called IRRemote would be fine enough as long as the directory you get from git is also called IRRemote and not Arduino-IRRemote?

This doesnt always workout / isn't always ideal. For example I was running CI testing with platformio but had to explicitly delete the RobotIRremote library for the testing to work.

If anything we might just rename the main header file... it would be slightly different to the name of the library but better than renaming the entire library?

marcmerlin, 2017-04-23:
I know that it works on arduino IDE if you have IRRemote/IRRemote.h (it will take it over RobotIRremote), but if that's not true for platformIO, then your suggestion to rename the .h file is probably the easiest, yes.

z3t0, 2017-04-23:
Progress!! 

https://github.com/platformio/platformio-pkg-framework-arduinoavr/issues/6#event-1053862405

closed 2020-10-12 by ArminJo