Arduino-IRremote/Arduino-IRremote #457: Cleaning up examples

z3t0, 2017-04-23:
What to fix:
- [ ] examples should only ever have one entry point i.e. IRremote.h not IRremoteInt.h

Files
- [ ] IRrecvDumpV2.ino
- [ ] IRremoteInfo.ino
- [ ] IRtest.ino
- [ ] LegoPowerFunctionsSendDemo.ino:

AnalysIR, 2017-04-23:
@z3t0 can you double-check that for :

IRrecvDumpV2.ino & IRremoteInfo.ino

..or am I looking at the wrong branch/files?

bengtmartensson, 2017-04-23:
Without having looked through all of them:

* Set the baud rate to 115200
* IRrevDump should produce output that is parsable by programs; in particular, no space between signs and the number, signs optional. Preferably on one (long) line. It should be possible to cut-n-paste it into your favorite program easily.
* Absolutely no includes of IRremoteInt.h.
* There must be a better way of solving the problem IRremoteInfo "solves". Among other things, should it be updated in the context of #437?

z3t0, 2017-04-23:
@AnalysIR I think the IRremoteInt.h might have been caused by an error on my side and only seen on my local copy?

Check https://github.com/z3t0/Arduino-IRremote/blob/dev/examples/IRtest/IRtest.ino as an example from the dev branch

z3t0, 2017-04-23:
@bengtmartensson 

> There must be a better way of solving the problem IRremoteInfo "solves". Among other things, should it be updated in the context of #437? 

Ideally I'd love to be able to bundle a python script or similar for this sort of thing, but I doubt that works in terms of portability

curiouspl2, 2017-06-08:
i've tried to test plain old RC5 remote and some DVD remote with
IRrecvDumpV2 
both get detected as 'unknown' 
if i switch to old IRrecvDump , they get detected ok (as RC5 and NEC) . 

RC5 remote is very old and certainly RC5 .

closed 2020-10-12 by ArminJo