Arduino-IRremote/Arduino-IRremote #458: Simplified API

z3t0, 2017-04-23:


z3t0, 2017-04-23:
@AnalysIR @bengtmartensson @ladyada @marcmerlin  thoughts?

AnalysIR, 2017-04-23:
I am not sure I fully understand your first paragraph. If you mean that there should be an example send & receive sketch for each supported protocol, then yes that would be a good idea.
By 'separate', if you mean protocol implementation should be 'isolated' from MCU specific stuff, then yes.

I suspect multiple sending & receiving is a very low % requirement.  Less than 0.001% (???)
IMHO both are usually handled better in HW, which also avoids complicating the FW.

The could be a case for IRrecv.on/IRrecv.off in addition to IRrecv.start/IRrecv.stop

start/stop needs to setup & enable/disable timer/interrupt...on/off just needs to enable/disable interrupt.

Not sure what the mask in sendRAW is for. But you might want to keep sendRAW the same and if needed create a new sendMask.

_**TYPO**: I suggest usung us vs ms for marks/spaces :)_

marcmerlin, 2017-04-23:
I didn't look deeply enough in the implementation to comment if the interfaces are the best ones, but generally splitting this up where each arch can define its own implementation of the function, would have made for a much less awkward port for me than #defines, or sometimes double #defines and other such tricks, so generally +1 from me on the general idea.
I'll ask the ESP8266 folks to comment on this.

bengtmartensson, 2017-04-23:
First of all, I wrote an IR library some time ago, [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino), see also the [API documentation](http://harctoolbox.org/infrared-apidocs/index.html). This demonstrates how such an API should be done, IMHO. Not to say that everything is perfect, but it is much more ""OO" than the current project. For me, it is somewhat hard to discuss improving IRremote without mentioning Infrared4Arduino.

> IRSend irsend(pin, khz); // Create send object on pin, need to figure out the pin part though...

1. Most IR sending implementation do not allow for the pin to be selected. However, some do, for example when using soft carrier. Question is how to handle invalid pin specification. Ideally, this should throw a compile time exception. 
2. The modulation frequency belongs to the signal to be sent, not the sender.

> irsend.mark(uint ms);
> irsend.space(uint ms);

I am unsure about this; it does not seem to be the right level of abstraction. The IrSend classes of Infrared4Arduino does not support these, just sending IrSignals and IrSequences.

> irsend.sendRaw(uint[] data, uint[] mask); // data is an array with the data, mask has a 1 for mark a 0 for space, might want a better implementation for this portion

As AnalysisIR said, mask is unclear, Also, C++ arrays are not Java arrays; you need to pass the length too. As I said previously, I want the frequency to belong to the signal.

> IRrecv irrecv(pin, khz); // Create recv object

What is the second argument good for? Selecting the receiver that has a TSOP**khz? Likely not... (There is some similar stuff in the /dev/lirc specification) Otherwise, using non-demodulating receiver, the modulation frequency is a measured parameter. But AFAIK, IRremote does not support non-demodulating receivers (TSMP58000 etc) at all. So???

> IRrecv.start(&uint[] recv_data); // pass a pointer to the place to save it

Not exactly a modern API, rather Fortran-ish. Infrared4Arduino uses dynamic memory allocation, which, some argue is a complete performance killer...

> All "protocol" implementations will be separate from the implementation of the library itself,

They should preferably be classes derived from some superclass. That way the user can #include the classes (s)he wants, without the present CPP juggling in the library itself.

Also, please consider migrating to platform-independent, C99 types like uint16_t etc. Presently, the data type used for the duration vectors is "unsigned int" which is 16 bits on some, 32 bits on other processors. Infrared4Arduino defines a `microseconds_t` and a `frequency_t`.

bengtmartensson, 2017-04-23:
Random requirements:

* There should be a function for just getting the raw microsecond data, without the decoding attempts (presentyl possibly only by fiddling with the preprocessor symbols).
* Presently, the [0] entry is junk (yes, I know what it means physically, still it is junk).
* The amount of time waited until the signal was considered ended should be the last entry.
* The timeouts at the beginning and at the end (see previous item) should be (runtime) configurable.
* Preferrable also the length of the received data (RAWBUF).
* User-level method of getting the durations in microseconds (presently must multiply by USECPERTICK)

z3t0, 2017-04-23:
> I am not sure I fully understand your first paragraph. If you mean that there should be an example send & receive sketch for each supported protocol, then yes that would be a good idea.
By 'separate', if you mean protocol implementation should be 'isolated' from MCU specific stuff, then yes.

Yes and yes.

> I suspect multiple sending & receiving is a very low % requirement. Less than 0.001% (???)
IMHO both are usually handled better in HW, which also avoids complicating the FW.

Good point. I will think about removing that then.

> start/stop needs to setup & enable/disable timer/interrupt...on/off just needs to enable/disable interrupt.

My hope was that there would be one interface for turning it on and off (including setup) because you could then cleanly integrate this library with others that share the same timer, also it is less confusing for the end user.

> Not sure what the mask in sendRAW is for. But you might want to keep sendRAW the same and if needed create a new sendMask.

~~The current sendRaw implementation alternates between mark and space~~ Now that I think about it your point makes more sense.. nevermind

z3t0, 2017-04-23:
@marcmerlin 

> but generally splitting this up where each arch can define its own implementation of the function, would have made for a much less awkward port for me than #defines, or sometimes double #defines and other such tricks, so generally +1 from me on the general idea.

I didn't actually think of that, but will look into implementing it. Currently though I meant the IR protocols being separated i.e. Sony, panasonic, NEC etc..

z3t0, 2017-04-23:
@bengtmartensson 
> Not exactly a modern API, rather Fortran-ish. Infrared4Arduino uses dynamic memory allocation, which, some argue is a complete performance killer...

My point is to try and get the library to be as scarce as possible in terms of resource usage. I will try out both implementations (dynamic and otherwise) and see how things go?

z3t0, 2017-04-23:
> Also, please consider migrating to platform-independent, C99 types like uint16_t etc. Presently, the data type used for the duration vectors is "unsigned int" which is 16 bits on some, 32 bits on other processors. Infrared4Arduino defines a microseconds_t and a frequency_t.

added to the OP

z3t0, 2017-04-23:
@bengtmartensson 

> Random requirements:

There should be a function for just getting the raw microsecond data, without the decoding attempts (presentyl possibly only by fiddling with the preprocessor symbols).
Presently, the [0] entry is junk (yes, I know what it means physically, still it is junk).
The amount of time waited until the signal was considered ended should be the last entry.
The timeouts at the beginning and at the end (see previous item) should be (runtime) configurable.
Preferrable also the length of the received data (RAWBUF).
User-level method of getting the durations in microseconds (presently must multiply by USECPERTICK)

All very good points, added to the OP

crankyoldgit, 2017-04-24:
> IRSend irsend(pin, khz);

Ditch the _khz_ as commented by others, it belongs in the sending routines, or if we want to expose the internals (mark,space etc) then we should create a setFreq(Hz) or similar. And, if we are going to re-work all of this, lets use SI-ish units. e.g. Hz. Even if it is mostly a placibo.

> irsend.start(); // enable sending

I was actually thinking of ditching start() from the ESP8266 library, as it really doesn't serve any purpose.
Not sure what it does on other variants. It's almost a no-op in reality. A user of the library should just start sending, not announce they are going to start sending IMHO.

> irsend.mark(uint us);
> irsend.space(uint us);

Please use uint32_t or larger for these. I think they should also be private, but can easily have my arm twisted. I'd suggest mark() to be private mainly because it is easy for an user of the library that doesn't know what they are doing to burn out the led potentially.
Also, add an optional arg of duty cycle `irsend.mark(us, duty cycle)` or put it elsewhere. e.g. setDutyCycle()

While here, can we standardise what mark(0) and space(0) really mean?
We had one arduino->esp8266 user who had an issue where the platform implementations where different. i.e. Arduino sent at least a single pulse, where as the software PWM implementation on the ESP8266 didn't. He was using mark(0) to produce a single pulse. Behaviour is undefiend.
Speaking of which, we moved away from using space(0) to turn off the LED as it isn't clear. We implemented a ledOff() instead.

> irsend.stop() // clean up use of timer, other applications can do what they want here.

Not something that is an issue on the ESP8266 implementation, but happy to stub it. Again, I think it should be private. Also, it should probably turn off the LED.

> irsend.sendRaw(uint[] data); // data is an array with the data

It should include what Hz & dutycycle to be set as, unless we set them elsewhere. In almost all ways, sendGC() is a better method/standard. As others have mentioned, we should include a length, or use a specific structure IMHO.

> IRrecv irrecv(pin, khz); // Create recv object

Not sure the value of setting the khz (again Hz pls! :) value here. It may give users an incorrect assumption that they can define their module receives at, or only receive that particular freq.

> IRrecv.start(&uint[] recv_data); // pass a pointer to the place to save it

We've toyed with the idea of making the buffer uint32_t[] to make the interrupt handler on the ESP8266 smaller and faster, at the cost of more memory. That's a trade off that suits us, as we have more memory available (IIUC) than the arduino platform, but we have distinct shortage of IRAM (CPU instructions held in ram, not data) where our interrupt handlers need reside. As we tend to have to co-exist with lots of other libraries, that space is v.small and a tight resource on our platform. Also, I'm considering making the buffer dynamically allocate-able as an option. The single size buffer design has a number of significant limitations.

> IRrecv.stop();

No comments. ;-)

> use platform independent types C99

Here here!! I've been moving our code base to that, only leaving the calling syntax the same as to not confuse users, and to give the appearance of similarity with your library. I'd love to move those over to platform independent too.

> There should be a function for just getting the raw microsecond data, without the decoding attempts (presentyl possibly only by fiddling with the preprocessor symbols).

Agreed, and there should be a way to copy or swap out the buffer to other memory, to allow reception to continue.

> Presently, the [0] entry is junk (yes, I know what it means physically, still it is junk).

Ack. It would be nice if it wasn't there, but the implementation cost to remove it might not be worth it.

> The amount of time waited until the signal was considered ended should be the last entry.

Ack.

> The timeouts at the beginning and at the end (see previous item) should be (runtime) configurable.

Ack. This is something I was going to implement (end time) for ESP8266 as several protocols (namely A/Cs) have monster gaps in them, and screw up the normal decoding flow. Hence, the desire for multiple capture buffers, and continuous (as possible) capture as an option. e.g. a ring buffer implementation.

> Preferrable also the length of the received data (RAWBUF).

Agreed, that's something we are also working toward in ESP8266 land.

> User-level method of getting the durations in microseconds (presently must multiply by USECPERTICK)

I'm not sure of the origin of the USECPERTICK, but I wish it would die in a fire. ;-) usec's are the appropriate unit (yeah, I know, not strictly an SI unit) we should be using for all things.

**General comments.**

I'm with @bengtmartensson on the OO point(s), we should be leveraging that more. I'm heading that way on our fork.

Things I've been working on for our upcoming v2.0 version is a more common code base. e.g. rather than re-implementing send routines in each protocol, try to have some common code/routine. e.g. We have a sendData() which does most of the protocol specs except for odd ball ones (e.g. the RC-?? family) and I'm going to be working on a common decodeData() routine as well.
[... cut ...]
We've had one request for multiple receivers on a single board. They eventually resolved their issue in hardware, but from a code/object point of view, we should support them. i.e. Move all of the irrecv dependencies into the object, don't rely on globals.

Oh, and one last thing, can we For The Love Of Zarquon, can we chose a [consistent code style](https://google.github.io/styleguide/cppguide.html) plz kthxbye?

There, that's my comments/suggestions for now. (You can _thank_ @marcmerlin for pointing me here :)

z3t0, 2017-04-24:
@crankyoldgit

> Ditch the khz as commented by others, it belongs in the sending routines, or if we want to expose the internals (mark,space etc) then we should create a setFreq(Hz) or similar. And, if we are going to re-work all of this, lets use SI-ish units. e.g. Hz. Even if it is mostly a placibo.

My reason for using kHz instead of Hz is that it requires less memory space to store the frequency.  Mark and space now have frequency parameters. 

> I was actually thinking of ditching start() from the ESP8266 library, as it really doesn't serve any purpose.

What I had in mind is that this would prevent users from getting unexpected behaviour when they try and use the same timers for other things eg LCDs. This way they explicitly reserve and free up the timer for the IRremote library.

> Please use uint32_t or larger for these.

How long can we expect a typical mark/space to be? Wouldn’t a uint16 suffice? (Note I haven’t used AC remotes so that may be different?) @AnalysIR?

>  I think they should also be private, but can easily have my arm twisted. I'd suggest mark() to be private mainly because it is easy for an user of the library that doesn't know what they are doing to burn out the led potentially.

I think it might be beyond us to protect the user from making that simple of a mistake? The reason for making these public is that the current protocol implementations will become their own sketches. For example instead of having the Panasonic code in the library we will have an example sketch called PanasonicReceive or similar, and the protocol implementation may want to use the mark / space directly. Though to be honest I’m not too convinced either now that I think about it, I would be open to making them private so whatever you think would be most appropriate :)

> Not something that is an issue on the ESP8266 implementation, but happy to stub it. Again, I think it should be private. Also, it should probably turn off the LED.

Agreed it should turn off the LED. Also don’t worry about any implementation at the moment, this is the plan for a major release and does not affect any existing PRs. Any changes that need to be made for existing ports/PRs can be handled when merging v3 upstream.

> Not sure the value of setting the khz (again Hz pls! :) value here. It may give users an incorrect assumption that they can define their module receives at, or only receive that particular freq.

Fixed. Though see my earlier comment about units. Also to be verbose KHz is SI, just not the SI *base* unit?

> We've toyed with the idea of making the buffer uint32_t[] to make the interrupt handler on the ESP8266 smaller and faster, at the cost of more memory. That's a trade off that suits us, as we have [...]

I am open to having two slightly different implementations on avr vs arm to best utilise the features of each. This might also resolve my earlier resposne about uint16 vs uint32.

> Also, I'm considering making the buffer dynamically allocate-able as an option. The single size buffer design has a number of significant limitations.

Could you explain this further? I am interested in either. My reasoning for the single buffer is performance, but I may be persuaded otherwise.

> use platform independent types C99
Here here!! I've been moving our code base to that, only leaving the calling syntax the same as to not confuse users, and to give the appearance of similarity with your library. I'd love to move those over to platform independent too.

Added to the OP.

> There should be a function for just getting the raw microsecond data, without the decoding attempts (presentyl possibly only by fiddling with the preprocessor symbols).

The receive data in this API is currently entirely RAW data, any decoding would be done in the sketch and be abstracted from the library itself. This is the best way to increase the customisability of the library for the end user.

> The amount of time waited until the signal was considered ended should be the last entry.
Ack.

Does ack mean yes or no :)

> Ack. This is something I was going to implement (end time) for ESP8266 as several protocols (namely A/Cs) have monster gaps in them, and screw up the normal decoding flow. Hence, the desire for [...]

Added.

> I'm with @bengtmartensson on the OO point(s), we should be leveraging that more. I'm heading that way on our fork.

I haven’t explored OO vs not OO on a micro controller, so this question may be naive. Is there any sort of memory usage hit from OO in micros?

> Oh, and one last thing, can we For The Love Of Zarquon, can we chose a consistent code style plz kthxbye?

Way ahead of you! We had #445 and will be using cpplint (google)

crankyoldgit, 2017-04-24:
> My reason for using kHz instead of Hz is that it requires less memory space to store the frequency. Mark and space now have frequency parameters.

Currently you seem to accept them (e.g. enableIROut()) as _int_ which I assume is int16_t on arduino (it is on ESP8266), a uint16_t which is the same size should hold freqs up to 65535Hz. I'm not saying there are IR protocols we might want to support above that, but it covers most of them. Plus, honestly we shouldn't really be storing the freq. should we? We are often only interested in the period time? Or perhaps the freq. is used in some other ways in other platforms?

> What I had in mind is that this would prevent users from getting unexpected behaviour when they try and use the same timers for other things eg LCDs. This way they explicitly reserve and free up the timer for the IRremote library.

Fair call. Our lib is totally software PWM at this point for sending. No interrupts used etc. Happy to keep a no-op stub for it in our lib.

> How long can we expect a typical mark/space to be? Wouldn’t a uint16 suffice? (Note I haven’t used AC remotes so that may be different?)

The inter-frame gap between commands often exceeds a uint16. e.g. the frame repeat time for NEC is 108ms (108000us). Our implementation supports the correct handling of repeats for the user, so it calls a space() for the appropriate gap if requested. The [Kelvinator A/C](https://github.com/markszabo/IRremoteESP8266/blob/master/IRremoteESP8266.cpp#L619) unit has a gap of approx. [40ms](https://github.com/markszabo/IRremoteESP8266/blob/master/IRremoteInt.h#L199).
I haven't come across any mark()s that come anywhere near requiring a uint32, but space() most definitely. Plus, we are concerned over two bytes on the stack?! A subroutine call will be more than that.

Re: private mark/space

I won't lose any sleep if they remain public. Just my suggestion/pref to make them private. The barn door has been open for a long time, so it is the question of do we bother closing it or is it too late.

Re: uint16 vs uint32 in the buffer.

I said we had toyed with the idea. Not really sure of the gain vs. differing implementations.
e.g. Going to uint32's for us would mean saving a few bytes of IRAM, and a handful less instructions executed per interrupt. (FTR, we interrupt on change of pin state) All at the expense of more memory and more code and more complexity. We'd have to post-process the capture buffer to do all the calcs we are currently doing in the Intr handler, before the user sees it. Currently, it's in the too hard basket, we have bigger fish to fry first. ;-)

> Could you explain this further? I am interested in either. My reasoning for the single buffer is performance, but I may be persuaded otherwise.

We have less overall memory pressure than you guys so we can consider trading off memory for functionality. One of the functions I want (i.e. I've personally hit) is trying to capture and decode multi-messages or v.long messages. e.g. The kelvinator a/c remote protocol had me capping out the 256 limit at one point. There was a bug at least in some implementations that didn't like/expect a buffer that big.
A simpler case is capturing Sony and the number of repeats etc. I'd love to be able to report to a user "hey we saw SONY 0xf00baa, and we saw it N more times."

The other reason is to keep capturing while decoding the current (previous?) signal. I've solved this in a way by making the decode routine optionally supply a second buffer, and copying the _live_ buffer into it.
This made (raw) dumping incoming packets have much less corruption. i.e. You couldn't re-enable capture (resume()) until the decode had finished as it used the _live_ capture buffer. If you enabled it before decode had finished, you stood the very likely chance of having data be corrupted while decoded.
With out that, the resume() often was performed mid-next-packet yielding odd/corrupted results.
This lead me to my current thoughts of having the irrecv'er copy out only the portion of the buffer when it detects a large-enough\<tm pat. pend.> gap, and store it in a linked list of smaller buffers which are _just_ the right size to fit the packet, rather than my current solution of a complete copy of the full size of the rawbuf.

The single buffer is fine for most applications, I was thinking of making it a call-time decision as to which implementation the user wanted. i.e. fast, small, fixed memory, but can't handle rapid multiple codes vs. not as fast, memory hungry, dynamic memory usage, but will handle multiple and/or long packets gracefully.

> The receive data in this API is currently entirely RAW data, any decoding would be done in the sketch and be abstracted from the library itself. This is the best way to increase the customisability of the library for the end user.

IMHO, We really want two levels of library/abstraction. One for the low(er) level stuff, used by power users and decoders/sendBlah etc; and another which is the typical end user. e.g. "I want to turn my Samsung TV on, and I just want to look up a hex code/Global Cache/etc string to do it." / "What command did my remote control just send?"

> Does ack mean yes or no :)
Yes. I didn't NACK ;-)

Re: OO vs. Non-OO memory usage.
No idea. We should get some data on that. However, I think it is more a problem for you AVR folk than for us in the "more bang for your $5" world. I am no compiler expert in how much overhead C++'s OO stuff adds. But hey, we're all using a C++ compiler, might as well use it. ;-)

> Way ahead of you! We had #445 and will be using cpplint (google)

\o/  \ o /   \ o / !!!!!1!eleven!!

crankyoldgit, 2017-04-24:
Just to clarify the initial point in my last reply, If you are concerned over storing the freq in Hz vs. kHz, you should have the arg/storage as a _uint8_t_ instead of _int_. That'd potentially save you a byte.

bengtmartensson, 2017-04-24:
>> I suspect multiple sending & receiving is a very low % requirement. Less than 0.001% (???)
>> IMHO both are usually handled better in HW, which also avoids complicating the FW.

> Good point. I will think about removing that then.

What is important is to clarify (and preferably enforce) what is supported and what is not, not to prohibit what one person finds unnecessary and uncool. For example, in Infrared4Arduino, multiple instantiation of the IrSendPwm class is prohibited by not having a public constructor for the class, instead a factory method is available, and can enforce the "Highlander principle" ("there can only be one...").

bengtmartensson, 2017-04-24:
> How long can we expect a typical mark/space to be? Wouldn’t a uint16 suffice? (Note I haven’t used AC remotes so that may be different?) @AnalysIR?

A "typical" duration is definitely < 65535 micros, the problem is the "a-typical" :-). I have [recognized the problem in Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino/issues/10). For example, when repeating a NEC1 signal, there is 96000 micros between the repeats. But doubling the required storage volume just for this is really a vast overkill. One possible solution would be to consider the numbers scaled by something, like 2, which would allow for up to 131071 micros. (This is really an argument for using a data type microseconds_t.) Another solution would be to have an extra variable for the last trailing space.  

All in all, I would argue for having typedefs frequency_t, microseconds_t, possibly also pin_t. 

>>    I'm with @bengtmartensson on the OO point(s), we should be leveraging that more. I'm heading that way on our fork.

> I haven’t explored OO vs not OO on a micro controller, so this question may be naive. Is there any sort of memory usage hit from OO in micros?

Every technology can have a bad implementation. I do not "think" a good OO  implementation "should" be more than marginally more resource consuming than a non OO. Suggestion: why don't you TRY Infrared4Arduino, and find out empirically if it is considerably larger or slower? Or possibly even better?
Test instead of guess!

bengtmartensson, 2017-04-24:
>> I think they should also be private, but can easily have my arm twisted. I'd suggest mark() to be private mainly because it is easy for an user of the library that doesn't know what they are doing to burn out the led potentially.

> I think it might be beyond us to protect the user from making that simple of a mistake? The reason for making these public is that the current protocol implementations will become their own [...]

It is not a matter of protecting silly people from doing silly things, it concerns having a consistent API with concepts/functions on the right abstraction level. A sender of IR signals should do that, and not expose mark() and space(). Also, a proper IR sequence has, IMHO, a "long" trailing silence, which is part of the signal, and "sent".  (Although I can accept an option for skipping the final wait-) This might eliminate the perceived need to expose space().

crankyoldgit, 2017-04-24:
> Also, a proper IR sequence has, IMHO, a "long" trailing silence, which is part of the signal, and "sent".

Agreed. It's part of the spec. Thus the sendFoo()s should support that, and that's what I'm doing on our lib.
I want it to be an emulation, not a _compatible_. Who knows what a receiving device might expect.

I sort-of had this as a problem for my home IR sending unit, where it would receive a series of IR commands for multiple devices, some of them requiring repeats. e.g. Sherwood equipment requires the NEC code + repeat. When I want to set the volume, I need to handle multiple repeat sequences (e.g. 30 or so) to emulate holding a volume button down.

crankyoldgit, 2017-04-24:
Re: larger than uint16_t for space() v.s. storage requirements.

I'm confused. Are we talking about the cost of calling space() with a uint32_t or are you talking about how you are going to store it in the receive buffer(s)? Or is it something else I'm missing?

bengtmartensson, 2017-04-24:
> I'm confused. Are we talking about the cost of calling space() with a uint32_t or are you talking about how you are going to store it in the receive buffer(s)? 

Also I am confused...
Assuming you mean me, I am talking about how to allow trailing spaces (sometimes called lead-out) to be > 65535 without doubling the storage space requirement.

crankyoldgit, 2017-04-24:
But storage where? (that's what I'm confused about)

In the sendBlah() code? i.e. the extra two bytes to do: space(70000) for a lead-out, vs. code to call space(1000) for example. Assuming a prototype of space(uint32_t us) for both.

Or the two (or more) extra bytes of stack space for the actual call/internals of the space() method?

Or are we talking about the ability to capture a space of 70000 usecs in the receive buffer and have two extra bytes per buffer entry?

bengtmartensson, 2017-04-24:
E.g. [here](https://gist.github.com/bengtmartensson/7eedcebd084fcee341a86e75fd54c75a)

crankyoldgit, 2017-04-24:
LOLWTFBBQ?!?!
Why on _earth_ store a protocol we have the encoding method for in an array format? Is that the extra storage you're talking about?

bengtmartensson, 2017-04-24:
Please adjust your attitude. I am sure the other participants understands my point.

crankyoldgit, 2017-04-24:
How about a compromise: We have:
```
void space(uint16_t us);
void spaceLong(uint32_t us);
```
If you want to save bytes in your array(s), we could define a mark(0) as a no-op, then you could effectively encode a v.large space in multiple uint16_t space()s.
e.g.: To do a lead-out of 108ms ...
[mark(650), space(650), mark(650), space(1300), mark(650), space(60000), mark(0), space(48000)]

crankyoldgit, 2017-04-24:
I'm clearly not understanding your point, hence why I keep asking. Your answer didn't exactly address my question(s) did it?

bengtmartensson, 2017-04-24:
if microseconds_t = uint32_t it takes up twice the space as when microseconds_t = uint16_t. Which is sometimes an issue, sometimes not. If writing reusable software, space is a priori an issue.

> mark(650), space(650), mark(650), space(1300), mark(650), space(60000), mark(0), space(48000)]

"Variable length encoding". Certainly one possibility.

crankyoldgit, 2017-04-25:
I think you might be conflating two orthogonal issues.
Having the calling proto of `void space(uint32_t us);` doesn't equate to a need to store all time values as uint32_ts. Yes, in an ideal world (less-resource constrained env.) we probably would/could/should etc.

I'm suggesting the minimal cost of upgrading the calling syntax of space() from uint_16t to uint32_t, which should realistically be in the realm of a few bytes temporarily on the stack etc, is worth it to accommodate safer/less complex code elsewhere[1]. *IF* however, we are quibbling over those few stack bytes, then I am at a loss. (Just to needlessly state the obvious, you can still call space() with a uint8_t value, and get the expected result. The compiler handles the sizing change for you statically.)

How a _user_ chooses to store their entire IR packet/message is up to them.
They obviously have a wide range of choices:

- A simple, yet storage expensive, low complexity _raw_ format. i.e. An array of maximum possible sized integers. For our current worst-case exercise, assume uint32_t[].
- A slightly more complex, yet smaller data footprint. e.g. A Global Cache format. (slightly encoded, way smaller uint16_t elements) [2]
- A specific, complex, high(est?) density, large code footprint, lowest data footprint. e.g. sendNEC().

We can't stop a _user_ from choosing the inappropriate one, nor can we provide them with a single optimal solution.

None of those _(user)_ implementation decisions really affects if space() is called with a uint16 or a uint32, nor (significantly) the memory it uses.

Obviously there is an entire spectrum of possible choices beyond those, each with their own trade-offs. As a library, we shouldn't be dictating which one they use/choose, but offer them multiple interfaces to help the choose & select more appropriate methods for their circumstances.

[1] Having extra code to check if we want to call space() with >=2^16, or handle those edge cases is ugly & complex, and thus invites errors/bugs/mistakes.
[2] It is left as a trivial exercise for the viewer to come up with a smaller data footprint variant of GC format that only stores timings in an array of uint8_ts.

bengtmartensson, 2017-04-25:
This is about internal storage of IR sequences. Yes, everyone with an intermediate understanding can suggest (countably) infinite ways of storing the duration data. The talk about `space(...)` is irrelevant; these should not be exposed to the user anyway. But, given the possibility of different internal representations, best thing to do is to encapsulate it in a class, like `IrSequence` in Infrared4Arduino. There, everything operates on instances of `IrSequence`s. 

> How a user chooses to store their entire IR packet/message is up to them.

You have a very different understanding than I and Infrared4Arduino. In my world, the user _is not free to select the internal representation_, (s)he constructs them using suitable constructors. Giving the user those "choices" seems to contradict the whole idea of "library".

bengtmartensson, 2017-04-25:
> Giving the user those "choices" seems to contradict the whole idea of "library".

Better: if you want to allow alternative representations, do it as inheritance from `IrSequence`, and allow alternative implementations to be implemented as subclasses of `IrSequence`. Main point is to encapsulate the internal storage,

crankyoldgit, 2017-04-25:
FWIW, we are really digressing from the OP. This is a beautiful rabbit hole.

> Giving the user those "choices" seems to contradict the whole idea of "library".

I'm going to agree to disagree with you on that.point/concept, and leave it at that.

Back to the original topic, can we please have the internal (or external, IDK) API allow space() to take a uint32 or similarly large value with whatever encoding, as long as it is performant and has low-ish complexity to encode. Pref. none.

I think we have concrete examples of > 2^16us values being needed to justify it. How we implement it, is another discussion for another time.

Also, on the original topic, I thought about it some more. There may be worth in adding default'ed arguments to the class constructors to enable setting things like the _default_ duty cycle (irsend), and buffer &/or size/timeout/assumed freq (irrecv). That's not to say they can't also have there own get/set methods or exposed variables either.

e.g. It would be nice to be able have a user tell the lib what freq. their IR demodulator is, so we can make assumptions based on that. Such as, here is the GlobalCache version of the 'raw' data for you, which we can't do without knowing/assuming the freq. etc. or allow for known bad reception of different freqs on that version of the demodulator. Low low priority IMHO. Just a thought.

I'd also love to be able to pass in a buffer to the class methods so we could unit test them, or similar mechanism. Hey, that's another advantage you AVR folk have over us in ESP land, we don't have a working unittest framework (that I know of) :-(
Passing in the buffer might a way of allowing the user to select the size at runtime too. ;-)

bengtmartensson, 2017-04-26:
#467 is very closely related to these topics.

ArminJo, 2021-02-03:
Tested space(uint32_t timeMicros): -> It costs 200 bytes flash. And the CPU can be (intransparently) locked by a library for a long time doing nothing.

closed 2021-02-03 by ArminJo