Arduino-IRremote/Arduino-IRremote #459: Memory footprint needs to be as small as possible.

z3t0, 2017-04-23:
Goal: 500 Bytes

z3t0, 2017-04-23:
Currently 9 bytes with an empty sketch using UNO

AnalysIR, 2017-04-23:
Needs to specified in terms of both RAM & flash.

I presume this is for AVR328p arduino as a pseudo reference platform. (vs 32 bit platforms)

...which raises the issue...Whats good for 8-bit (low resource) isn't always ideal for 32-bit (lots more resources).

> Currently 444 bytes with an empty sketch using UNO

If the sketch is empty, the compiler may have dropped out most of the code?

You may be better referencing the size increase between a basic sketch using serial, millis, micros etc and some reference IRremote sketches (with/without Serial?)
- Rx, size for only decoding NEC.
- Tx, size for only sending NEC.
- Tx/Rx, size for sending & decoding NEC.

It could also serve as an ongoing baseline comparison for checking the impact of any future changes.

closed 2020-10-17 by ArminJo