Arduino-IRremote/Arduino-IRremote #461: API documentation using Doxygen

bengtmartensson, 2017-04-23:
Writing API documentation is a major TODO, as I see it. I would here like to recommend [Doxygen](http://www.doxygen.org), which generates the API documentation from Javadoc like comments. For an example, see [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino), with the produced documentation available [here](http://harctoolbox.org/infrared-apidocs/index.html). The documentation is generated at the developer's PC. and (for example) uploaded. The above mentioned Infrared4Arduino contains the Doxygen runs in the Makefiles.

I can contribute the main configuration, the `Doxyfile`. Documentation for the individual classes/function do not need to be written at one time, also withoth the comment the output is usable.

Writing a "book" on the API never works: either it will never be finished, or is obsolete by the time on finishing... (See IRLib for an illustration.)  Generated API docs is the way to go (agile).

bengtmartensson, 2017-04-24:
Here is [doxygen.zip](https://github.com/z3t0/Arduino-IRremote/files/950894/doxygen.zip), containing

1. A possible Doxygen file (the one from Infrared4Arduino, slighly adapted)
2. Output from a run. To read, just open `api-doc/index.html` in your favorite brower.

bengtmartensson, 2017-08-27:
Just PRed an update to README.md explaining how to invoke doxygen.

What remains to do (if desired :-) ) is to integrate the doxygen run into the CI process and to make the generated API documentation available for the user in, e.g. the Wiki. Also, this will make some existing parts of the Wiki obsolete. Thoughts?

z3t0, 2017-08-27:
I'm okay with overwriting the wiki as that's been a work in progress on my local git for a while.

Thanks,
Rafi Khan

z3t0, 2017-09-07:
I'm not really sure how to use CI for doxygen (will look into it) but for the time being it's an easy matter for me to manually do so for each commit that lands in master

bengtmartensson, 2017-09-11:
After all, I do not think it is a good idea to put automatically generated content in a Wiki; it is simply not intended for that. Instead, it might be better to put it on a GitHub page. As a demonstration, I did this for my fork of the project, [see this](https://bengtmartensson.github.io/Arduino-IRremote/)

bengtmartensson, 2020-06-25:
Doxyfile and rudimentaryAPI documentation is present in the current master, so I consider the issue as implemented. Remains, if desired, to publish, possibly automatically generate and publish as part of the build/release process. And, of course, improving the documentation ;-). Closing this issue.

closed 2020-06-25 by bengtmartensson