Arduino-IRremote/Arduino-IRremote #462: Remove LED blinking

bengtmartensson, 2017-04-24:
Is the LED(-13) blink really a core functionality of the library? Or is it a debug feature that never was taken out? 

Disadvantages: 
* increases porting (and cleanup :-)) effort. 
* Locks up a LED that the user might like to use for some other purpose
* Can be annoying in deployment

Advantages:
* might be useful by initial debugging.
' Looks cool if you like it

Suggestion: take it out.

z3t0, 2017-04-24:
> Suggestion: take it out.

Agreed.

z3t0, 2017-08-17:
I have reopened this issue after considering that the LED might actually be useful to the end user. As such I would suggest rather than removing it we address the issues.

For example, it can be disabled on platforms that do not support it and we can add a relevant section in the documentation that explains how to disable the LED for advanced users.

bengtmartensson, 2017-08-22:
This issue has been resolved (by rejection). The issue thereby can and should therefore be closed.

commit a406f3b9 (2017-08-13): Removed blinking led

It is difficult to maintain this over different architectures and also
not entirely that useful.

Fixes #462
closed 2017-08-22 by z3t0