Arduino-IRremote/Arduino-IRremote #464: Make source file layout compliant with Arduino library standard

bengtmartensson, 2017-04-24:
There _is_ actually an [Arduino library specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification). According to that, the sources should be a in a subdirectory names `src`, and the non-public (not exported) header files in sub-directory/sub-directories of that directory. 

This should possibly replace #375 (which I am personally find very unsatisfying).

Not really urgent.

z3t0, 2017-08-13:
Is there an official reference document somewhere to follow, or should I just conform to the layout of official libraries

z3t0, 2017-08-13:
This might be what we are looking for: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

bengtmartensson, 2017-08-13:
That was the document I linked to in the original post ;-).

z3t0, 2017-08-13:
Ah just saw that... oddly the original post was blank when I originally saw it, just the title was visible

z3t0, 2017-08-13:
Please review the commit 635e8a9 and then we can close this issue.

bengtmartensson, 2017-08-13:
The most important thing I believe is not the `src` directory but the hiding of the "private" includes (`boarddefs.h, IRremoteInt.h`) in a sub directory. I just read the library guidelines again, and I see that there is now an alternative method. (I cannot recall that it was there when I wrote the OP.) 

> includes - (available from IDE 1.6.10) a comma separated list of files to be added to the sketch as #include <...> lines. This property is used with the "Include library" command in the IDE. If the property is undefined all the headers files (.h) on the root source folder are included.

So I ask you to implement one of these methods. I think the sub directories is the preferred one: It works for IDE < 1.6.10, and it is more obvious to the user that some includes are public, some not. The drawback is that all the corresponding include statements need to be changed.

z3t0, 2017-08-13:
My preference is for the first method where we use subdirectories to hide
the libraries own headers, that way the top level headers are much more
accessible and there is a clear distinction in the software design.

Thanks,
Rafi Khan

bengtmartensson, 2017-08-21:
Has been fixed by PR #507. I take the liberty to close.

commit 24dd9401 (2017-08-20): Moving private includes to sub-directory,  #464 (#507)

* Add missing "paragraph" to library.properties.

* Move not-to-be-exported includes to subdirectory.
Remove #include "IRremoteInt.h" from a number of files since it
now does not work any more, and since IRremote.h #includes it already.
This resolves #464.

* Remove #include <IRremoteInt.h> from examples, too.

* Make ir_Lego_PF_BitStreamEncoder.h exported (again).

* Revoked fe24095 (paragraph) to keep 1-1 correspondance to #464.
closed 2017-08-21 by bengtmartensson