Arduino-IRremote/Arduino-IRremote #466: Having trouble with Pronto

beaglesparks, 2017-04-25:
**Board:** Nano
**Library Version:** 2.2.3
**Protocol:** Pronto

char DISH_GUIDE[] = "0000 0048 0000 000b 0016 021a 0016 009d 0017 00fa 0017 009b 0016 009b 0017 009c 0017 00fb 0017 00fb 0017 00fb 0017 00fb 0017 00fa";

...

       irsend.sendPronto(DISH_GUIDE, false, false);

I get the following error

undefined reference to `IRsend::sendPronto(char*, bool, bool)'

collect2.exe: error: ld returned 1 exit status

What am I doing wrong?

Thanks

z3t0, 2017-04-25:
Hmm might need more information.. no idea what collect2.exe is?

beaglesparks, 2017-04-25:
me either.  i've searched but can't find the reference.  Let's ignor it for now.  am i declaring DISH_GUIDE correctly? Also am i calling irsend.Pronto correctly?

beaglesparks, 2017-04-25:
I found collect2.exe.  it seems to be a secondary error related to the IDE https://forum.arduino.cc/index.php?topic=310950.0

bengtmartensson, 2017-04-25:
`collect2` is the main program of the loader. The original error messages means that the loader cannot find the said functions. 

The loader cannot find the function because no such function is not defined. In Pronto.cpp there is a function (line 73)

    bool sendPronto (char* s,  bool repeat,  bool fallback)

but defined that way, it is a global function, not a member function of IRsend (`IRsend::sendPronto`).

Quickly browsing through Pronto.cpp I see a number of problems...

z3t0, 2017-04-25:
Yep. I had deleted Pronto from the dev branch as well realising that either it is very poorly written or just written for a very specific use case a long time ago. That's about all I can offer at the moment though...

beaglesparks, 2017-04-25:
I have a Dish DVR i want to control and pronto codes for it.  do you have suggestions on what i need to do to try to control it? I can only find pronto codes for it, nothing else

bengtmartensson, 2017-04-25:
You can translate the Pronto form to raw microseconds using e.g. IrScrutinizer. The given one corresponds to 

+382 -9345 +382 -2727 +400 -4342 +400 -2692 +382 -2692 +400 -2710 +400 -4360 +400 -4360 +400 -4360 +400 -4360 +400 -4342

(Possibly you should dispose ot the signs.)

bengtmartensson, 2017-08-13:
Suggestion: delete irPronto.cpp. I do not think anything needs it.

z3t0, 2017-08-13:
Reverted changes check #502

closed 2017-08-13 by z3t0

seltix5, 2019-12-17:
> You can translate the Pronto form to raw microseconds using e.g. IrScrutinizer. The given one corresponds to
> 
> +382 -9345 +382 -2727 +400 -4342 +400 -2692 +382 -2692 +400 -2710 +400 -4360 +400 -4360 +400 -4360 +400 -4360 +400 -4342
> 
> (Possibly you should dispose ot the signs.)

hello! how do you send this raw microsecons with IRremote? thanks!

AnalysIR, 2019-12-17:
**Why add this out-of-context request to a post abput PRONTO?**

So the short answer is work thru the examples provided with the library.

seltix5, 2019-12-17:
Because I try to use Pronto to send remote codes but it did not work. As your suggestion I try to use the example code to send the raw signal but I did not had success either. My only working finding at the moment was using the Pronto hex code and irscrutinizer to export in arduino/ir remote format but I wold like a automatic solution working directly frok the Pronto code or using a online converter...

bengtmartensson, 2019-12-17:
Huh? You know how to generate IRremote code from Pronto, and that is not "automatic enough"... What is more automatic with an online converter? What do you want, really?

seltix5, 2019-12-17:
i just want a way to automate the conversion... without manual conversion. with online converter I can create a script to automate the process without the need of using a local tool

bengtmartensson, 2019-12-17:
So what is then your input file format? Pronto Hex signals (i.e. collections of those, with names) normally come in machine readable format (for example on Remote Central).

seltix5, 2019-12-17:
for my development stage my current input is the irplus WAVE app for android from where I use the export method to retrieve the pronto hex list codes (i'm using the app because I can confirm if the code is working). After development i wold like to link it with online code databases to retrieve the codes but the online sources always uses the pronto format

commit 563c82c6 (2017-08-13): Removed pronto "protocol"

Fixes #466