Arduino-IRremote/Arduino-IRremote #467: Use C only? (not OOP)

z3t0, 2017-04-25:


bengtmartensson, 2017-04-25:
First, you do not mean really deploying the C compiler, using C type-unsage linking etc, do you? I do not think the Arduino platform supports that, although I might be wrong (i.e. there may be a dirty trick...)

I think it may be a good idea to consider having a certain core in non-OO, "Fortran-type" style, lightweight, no dynamical memory allocation, etc. If this really will be a "library IRremote-core", or just a conceptual "level" I leave open right now. I think the number of public functions can be reduced to two, as discussed in what follows.

I would suggest using the following typedefs/enums, for encapsulaton

    typedef *** microseconds_t;
    typedef *** frequency_t; // Semantic to be determined (Hz or kHz :-))
    typedef *** pin_t; 
    enum {
          timeout,
          invalidPin,
          hardwareBusy,
          ....
     } ErrorCode;

For sending, there is the public function

    ErrorCode sendIrSequence(pin_t pin, frequency_t frequency, const microseconds[] data, size_t length);

which will work almost like the present  `irSend.sendRaw(....)`. It returns a statuscode, for example if the given pin is not valid, or the hardware is busy or not capable of fulfilling the request. (in the absence of exceptions!) This should normally be called with an even number of data points (last being the ending, "long" space), but also calling with an odd number of data points should be allowed.

There may be a version with defaulted (platform dependent) pin.

For receiving, there should be one function

    ErrorCode receiveIrSequence(pin_t pin, frequency_t *frequency, size_t *length, microseconds_t *data, startTimeout=defaultTimeout, endTimeout=defaultTimeout);

* frequency is an out-argument, for the receivers that deliver it (presently IRremote does not do that).
* data is an array allocated by the user,
* lenght is on input the length of the allocated data, by output the number of data points (including final silence)
* If no signal has been detected for the time denoted by startTimeout, the function should return with the return code `timeOut`.
* The final time interval of required silence is configurable as the endTimeout argument.

Having a core of just these two functions, everything else should use just these two. The mid- or high-level functions/classes (which can be OO or not) should build on this core. The user should thus be discouraged to use the core functions directly.

Aside: The Google C++ guide specifies input arguments before output arguments, which makes the C++ default argument mechanism useless... The Arduino style guide discourages pointers, replacing them by references is prohibited by the Google guide, which states that references should be `const`. Sigh... 

Being the author and maintainer of Infrared4Arduino, I would also be very interested to be able to replace all (most?) the explicit hardware access therein by using such an "IRremote-core".

Thoughts?

bengtmartensson, 2017-04-26:
Let me modify the "only two functions" statement a bit. If there are several implementations that should be offered to the user, it may be a good idea to have them as functions with different names. For example, there can be the receiving functions

    ErrorCode receiveIrSequenceSampler(pin_t pin, size_t *length, microseconds_t *data, microseconds_t startTimeout=defaultTimeout, microseconds_t endTimeout=defaultTimeout);
    ErrorCode receiveIrSequenceInterrupt(pin_t pin, size_t *length, microseconds_t *data, microseconds_t startTimeout=defaultTimeout, microseconds_t endTimeout=defaultTimeout);
    ErrorCode receiveIrSequencePoll(pin_t pin, size_t *length, microseconds_t *data, microseconds_t startTimeout=defaultTimeout, microseconds_t endTimeout=defaultTimeout);
    ErrorCode receiveIrSequenceCapture(pin_t pin, size_t *length, microseconds_t *data, microseconds_t startTimeout=defaultTimeout, microseconds_t endTimeout=defaultTimeout);

(note that only the last one has `frequency`). Likewise for the sending functions.

Should there possibly be one function `sendIrSequenceSoft` using soft carrier, available on _all_ platforms (even those supporting hardware PWM), and `sendIrSequenceHard` available only where implemented?

crankyoldgit, 2017-04-26:
re: style guide.

argument order:
_"This is not a hard-and-fast rule. Parameters that are both input and output (often classes/structs) muddy the waters, and, as always, consistency with related functions may require you to bend the rule."_

[const references](https://google.github.io/styleguide/cppguide.html#Reference_Arguments): They seem pretty reasonable to me. I wouldn't like a function to change a pointer argument I passed to it. Would make memory leaks galore etc.

bengtmartensson, 2017-04-26:
@crankyoldgit :
First of all, _output arguments are considered evil_ in modern OO programming, see e.g. Martin: "Clean Code", page 45.  Results should be delivered as value of the function.  Some languages allow multiple output arguments (Matlab, Lua, Python,...), C/C++/Java etc do not. So in these cases we need to define a class (struct) that holds the output stuff.  Assuming that @z3t0  wanted a "Fortran-ish" solution with output arguments, I presented one such, although I subscribe to "output arguments are evil". 

Question is if we are to adopt a guideline that will be broken immediately, or simply does not make sense.

> const references: They seem pretty reasonable to me. I wouldn't like a function to change a pointer argument I passed to it. Would make memory leaks galore etc. Edit: The pointer changing that is, not the value it is pointing to obviously.

Your argument is an argument against output arguments, not of preferring pointer over references. 

I still wait to see the rationale for allowing `foo(int * outputarg)` but not `foo(int& outputarg)`.

crankyoldgit, 2017-04-26:
FYI, I wasn't commenting on your implementation etc.
I'm just not sure you were painting their style guide in the best light.

I agree, it's best to use the return value of a function to return a value. However, that's not always efficient/practical/useful. The style guide doesn't say you have to use output args, but if you do, put them at the end. Seems sort of reasonable to me. shrug.

Re: rationale. I don't know. I can't speak for them. One of things they list _"Defining a parameter as reference avoids ugly code like (*pval)++. Necessary for some applications like copy constructors. Makes it clear, unlike with pointers, that a null pointer is not a possible value."_ sounds like a good one to me.

But hey, before you throw the baby out with the bathwater, remember, [it's a guide, not a rule](http://imgur.com/gallery/mSHi8).
Use the bits you want, and if you've got good reasons, ignore the bits that don't fit. There is no reason you guys can't have your own super-set of guides on top of theirs. Saves having to re-invent the wheel etc.

bengtmartensson, 2017-05-12:
Reverting to the original question. I am considering if it would be a good idea to have and maintain 

## TWO versions of the library

(or possibly "two libraries")? One "efficient" and lightweight, basically "C", no dynamic objects, no String's, etc, another with OO design, API, and cool features. My previous postings can be said to investigate the possibility of having a common basis for both versions.

z3t0, 2017-06-08:
I agree with the idea of a core library that houses all board definitions and will be useable by other libraries that wish to build on top. 

I'll hash out a simple implementation over the next few days and then we can all point the errors of my ways and make it something worth releasing. 

I'm not at all concerned about style of code and am open to any suggestions you think would be the best. My only concern here is performance and reliability.

closed 2020-10-12 by ArminJo