Arduino-IRremote/Arduino-IRremote #469: Pin used for sending on a Arduino MEGA

feelgoodgirl, 2017-05-03:
Heyy,

Currently i am using this library. This works great. It took a while before I realized that for Arduino MEGA pin 9 it was used for sending.

This is why my question is. Because when you look in the list it is possible to use Arduino Mega pins 5, 6, 9, 11, 46. However, if I get another pin than 9, it does not work. Is this correct or I am doing something wrong?

I am a beginner in programming. I'm trying to understand what I'm programming.

I am looking forwared to your respons!

z3t0, 2017-05-03:
Pin 9 is the default. You change the pin by commenting an unconsenting two lines in one of the library header files.

I'm out now but will give specific instructions when I get a chance

feelgoodgirl, 2017-05-04:
Thank you for your comment. I have looked at this part in the different libraries and have indeed found this in the 'boarddesfs.h'.

What is the reason you did not choose to adapt this to the program? For example, you just have to indicate which pin you use.

z3t0, 2017-05-05:
The main issue we have is the way compilation works. First the library is compiled then the program. 

This means the code can't affect the library and so change the timer. 

I am open to suggestions on how to improve this!

feelgoodgirl, 2017-05-05:
I am way to new at Arduino to give some advice/suggestions. Thank you for clarification. my infrared project works fine. Now I'm going to try bluethooth.....

z3t0, 2017-05-05:
Great, closing the issue for now. Feel free to reopen if you still need help with this.

closed 2017-08-22 by z3t0