Arduino-IRremote/Arduino-IRremote #471: Timer selection

DrDiettrich, 2017-05-09:
Currently the library has to be modified, in order to select a specific timer. That's bad because such a modification will affect all projects, or the entire library has to be copied into a specific project before.

If dedicated classes are declared, each bound to a specific timer, the user can select the timer to use in its project file, by instantiating an object (or more) of the specific class(es), like
IRsend1 irsnd1; //using T1
IRrecv2 irrcv2; //using T2

This can be accomplished by multiple #includes of the headers and code, each with a different IR_USE_TIMERx or similiar, so that no duplication of the class declarations etc. is required. Then all classes are compiled at once, but the linker will include only the really used one(s) into a project.

A small number of modifications to the current code are required, to allow for proper compilation of multiple inclusion of the same header and source files. If e.g. TIMER_TO_USE is set to the timer suffix, the class names can be constructed like
#define IRrecv IRrecv ## TIMER_TO_USE
#define irparams irparams ## TIMER_TO_USE
because macro expansion is non-recursive.

I'm willing to contribute the changes, e.g. in a new branch.

z3t0, 2017-05-10:
I like this idea! I'll add it to the 3.0 release todo list. I wouldn't bother doing a full implementation at the moment as there are going to be breaking changes, but if you want you can take the initiative to plan out how this implementation would be structured and how the end user would interact with the multiple options etc.

bengtmartensson, 2017-05-10:
Let me add my 0.02 cents...

> Currently the library has to be modified, in order to select a specific timer. That's bad because such a modification will affect all projects, or the entire library has to be copied into a specific project before.

Yes. Requiring the user to modify the source is in the strict sense violating the whole concept of "library".

> This can be accomplished by multiple #includes of the headers and code, each with a different IR_USE_TIMERx or similiar, 

To my knowledge, using the standard Arduino IDE we all know, love or hate, it is not possible to control the include files that the library uses, nor to control the C preprocessor definitions used in the compilation of a library. If I am wrong, please correct me. Assuming that is not the case, also with your modifications, the library principle is violated.

What I would like to propose instead is this; first considering sending only. I suggest (e.g.) replacing the constructor(s) of IRsend by a factory function in IRsend, in the versions

    public:
         static IRsend* newIRSend(); // version with default timer/pin
         static IRsend* newIRSend(unsigned timerToUse); // pin is determined from configuration
         static IRsend* newIRSend(unsigned timerToUse, unsigned pin);

    private:
         IRsend(unsigned timer, unsigned pin); // private, the user should use the factor instead

Note that some configurations (a softcarrier implementaton, the Due, possibly others) allows the selection of different pins, thereby the third version. If the user is requesting an invalid combination, ideally this should be determined at compile time, otherwise, NULL should be returned (better would be, at least IMHO, to throw exceptions, but they seem to unsupported in the Arduino environment). The factory functions can also cleverly control to what extent multiple instances are allowed, returning NULL by failure. It should also cover using the soft carrier implementation (timer = -1?).

The file `boarddefs.h` (or some successor thereof) should contain the info on what timer arguments are allowed, and the possible pins for the different timers.

DrDiettrich, 2017-05-10:
Dynamic objects are a bad idea with most 8-bit AVRs. Even if that were a traditional OOP approach, static objects are preferable. As already mentioned, only the really used static structs and methods are bound to the final binary, so that the user has full control over the memory footprint.

I'm not sure about the ISRs yet, these may deserve special handling (weak?) for not blocking unused vectors. If nothing helps, I'd try to implement these in a single module, that is #included only in the project file. This module again can #include further files (see my initial code snippet), no need to force everything into a single file. This approach allows the user to specify the intended timer before including that module, and no further modules are compiled separately, out of user control.

Here is the [solution](https://forum.arduino.cc/index.php?action=dlattach;topic=475861.0;attach=211098) that inspired me.

DrDiettrich, 2017-05-12:
After thinking a bit more about the issue, I don't think that multiple senders or receivers at the same time are really useful. So what remains is a user-configurable IR_USE_TIMER. This can be accomplished as outlined just before, so that the entire code can be #included into the project file, and only one instance of IRsend, IRrecv and ISR is created for the indicated timer. The user then simply adds code like this to the project:

#define IR_USE_TIMER3 //if not defined, use the default timer as before
#include <IRremote.h> //definitely implements the classes, here for timer 3
IRsend irsend();
IRrecv irrecv();

bengtmartensson, 2017-05-13:
> I don't think that multiple senders or receivers at the same time are really useful.

There a commercial products supporting several transmitters, see for example [Global Caché](https://www.globalcache.com/products/). But it is certainly the most basic requirement.

> what remains is a user-configurable IR_USE_TIMER

... without editing the files belonging to library, which would invalidate its "libraryness". 

> This can be accomplished as outlined just before, so that the entire code can be #included into the project file

Using the established code organization (implementation in *.cpp, public interface in *.h), I argued in my last post that this is not possible using the current Arduino IDE.

DrDiettrich, 2017-05-18:
When IR is reflected, multiple senders or receivers don't make sense physically. Multiple senders can be connected to the same output pin, no need for special software support.

The IDE also doesn't support project specific global defines, which are the traditional way to configure libraries. That's why I suggested a workaround.

closed 2017-05-29 by DrDiettrich