Arduino-IRremote/Arduino-IRremote #473: Is it possible to send and receive with one Arduino UNO board?

darkengine, 2017-05-22:
Hi, I want to implement a feature like this. IR send an receive with one UNO R3 board. Green LED on when receive data(don't care what it is); red LED on when receive nothong. The IR receiver can receive data from TV remote with only "irrecv.decode" in loop() and output to serial port. However when I add “ irsend.sendNEC(0xFF02FD, 32);” the receiver stop working with TV remote.

So I create an issue to ask wether send and receive in one board is possible. And how to do that if yes with the first question. Many thanks.

**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** NEC

**Code Block:**
#include <IRremote.h>

int RECV_PIN = 11;
int GREEN_PIN = 7;
int RED_PIN = 8;

IRrecv irrecv(RECV_PIN);

IRsend irsend;

decode_results results;

void setup()
{
  // setup LED pins
  pinMode(GREEN_PIN, OUTPUT);
  pinMode(RED_PIN, OUTPUT);
  green();
  Serial.begin(9600);
  // In case the interrupt driver crashes on setup, give a clue
  // to the user what's going on.
  Serial.println("Enabling IRin");
  irrecv.enableIRIn(); // Start the receiver
  Serial.println("Enabled IRin");
}

// green light means no car here
void green() {
  digitalWrite(GREEN_PIN, HIGH);
  digitalWrite(RED_PIN, LOW);
}

// red light means park position been occupied
void red() {
  digitalWrite(GREEN_PIN, LOW);
  digitalWrite(RED_PIN, HIGH);
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
    green();
  } else {
    red();
  }
  delay(200);
  // send pin has been set to PIN_3 in header, so connect it to pin_3
  irsend.sendNEC(0xFF02FD, 32);
  delay(100);
}

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2017-05-22:
tr this after sending each  signal...(Rx is turned off every time you send a signal)
`irrecv.enableIRIn(); // Start the receiver`

LEDs
You really only need one LED. say blink the red when a signal is received and off otherwise. Seems to achieve the same functionality as 2 LEDs

darkengine, 2017-05-23:
Thanks for the advice. However I found it impossible to do that after reading the source code. When sending data:

void  IRsend::enableIROut (int khz)
{
// FIXME: implement ESP32 support, see IR_TIMER_USE_ESP32 in boarddefs.h
#ifndef ESP32
    // Disable the Timer2 Interrupt (which is used for receiving IR)
    TIMER_DISABLE_INTR; //Timer2 Overflow Interrupt
    ... ...
#endif
}

It means there's no chance for receiver to be waken up by interrupt at the same time.

AnalysIR, 2017-05-23:
Are you using the master branch?

darkengine, 2017-05-23:
Yes, the master branch.

AnalysIR, 2017-05-23:
OK, which file contains the code you posted above?
_(I did a quick search in the master branch & couldnt find it)_

AnalysIR, 2017-05-23:
> So I create an issue to ask wether send and receive in one board is possible.

If you mean that you want to send & transmit with the same arduino using both rx/tx at the same time. this generally doesn't make sense as the sending will usually come back to the receiver immediately (unless very well isolated).
For that reason, most IR implementations, will disable Rx before sending begins & re-enable rx after Tx is completed.

Maybe if you can provide more details about your application/requirments, I can provide a better answer.

z3t0, 2017-05-24:
Also, this might be silly but if you are using the same protocol for each sending and receiving whats to stop the sending from being intercepted by the receiving

darkengine, 2017-05-25:
Through my requirments, it needs some multi-threading to implement. There's too far from an UNO's ability. So I'll close the issue. Thanks all.

closed 2017-05-25 by darkengine

moshfeu, 2018-03-10:
@AnalysIR  Thanks!