Arduino-IRremote/Arduino-IRremote #474: Integrate IRremote-SIL?

bengtmartensson, 2017-05-23:
[IRremote-SIL](https://github.com/bengtmartensson/IRremote-SIL) is a small SIL (Software-in-the-Loop) test rig that allows to compie/run/test/debug the `IRsend::send` functions on the host, i.e. without flashing them onto the Arduino. It does not require any changes to the IRremote sources. It is intended for developers of the library, and authors of new protocols.

Cf. #452 

Can/should this be "integrated"? A pointer in the README might be useful.

z3t0, 2017-05-24:
Seems like a good solution. I definitely think we should integrate this as it would help clear up many issues with protocols and their implementations differing.

ArminJo, 2020-10-19:
Interesting software!
But for a new protocol I start with a similar one and change the timing and test it on real Hardware.
Most protocols are now using
`sendPulseDistanceWidthData(NEC_BIT_MARK, NEC_ONE_SPACE, NEC_BIT_MARK, NEC_ZERO_SPACE, tCommand, 16, false);`
or similar, and thats it.
And for other coding schemes we should deliver a send... function like above which makes generating new protocols more foolproof.

closed 2020-10-19 by ArminJo