Arduino-IRremote/Arduino-IRremote #475: Help adding Lutron protocol

BrizzzUK, 2017-05-28:
I've been trying to add the Lutron Dimmer protocol to the library but am struggling with how to convert the protocol timings supplied by Lutron to the #DEFINE entries needed in the ir_Template.cpp.

This is the protocol definition provided by Lutron:

IR Carrier Frequency:			40.0 kHz
Duty Cycle:				40%
Single Bit Time:			2.288 milliseconds
Baud Rate:				437 bps
Command Length:			36 bits
Command Duration:		82.368 milliseconds
Logic One:				Presence of IR modulated at 40.0 kHz
Logic Zero:				Absence of IR
Transmit Order:			Transmit the most significant bit first
General Function:			IR code is transmitted while a button is held down
Timeout Function:	Timeouts may not occur until at least seven-seconds of continuous IR transmission has taken place.

I know there's an issue with >32 bits but all of the Lutron codes begin with 1111 so I guess this could be hardcoded and sent as a header in the protocol.

The full protocol definition is in the attached pdf. Can anyone shed some light to help me move forward?

[Lutron.pdf](https://github.com/z3t0/Arduino-IRremote/files/1034395/Lutron.pdf)

AnalysIR, 2017-05-28:
This is a very unusual IR protocol.....
...logic 1 is indicated by mark & 0xF would be 4 marks combined...similarly for logic 0 but with continuous space to indicate multiple 0s.

I suspect you night be better using the RAW features for this protocol.

However, if it always starts with 1111, you could use that as a pseudo 'header' (which it probably is) and just treat the remainder as a 32 bit unsigend int(long).

jxwolstenholme, 2018-09-07:
Hi @BrizzzUK - wondering if you ever found a solution to working with the Lutron protocol?

Having a really hard time with getting my Grafik Eye to read any signals that I send - even in RAW format.

AnalysIR, 2018-09-07:
The nature of this protocol makes it very tricky to recreate.

Suggestions:
- Make sure you have a good IR emitter. Bring it closer for testing and point the LEd directly at the target receiver.
- Try recording with with our sketch, which will give better results than stock IRremote. [link ->](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)
- Keep the remote about 1->2m away from the receiver when recording & use a 40kHz receiver if available (otherwise 38kHz should do)
- Finally when sending the raw signal, make sure that every individual ,ark/space length is rounded to the nearest 2288 uSecs (calculated from 437bps)

If you still have issues post an example signal that you recorded and also the one that you tested, plus the model of IR receiver used & a description of your IR emitter circuit.

jxwolstenholme, 2018-09-07:
Hi @AnalysIR 

I don't actually have the original remote for my system (Grafik Eye) so am trying to convert the pronto hex codes provided in the document referenced above to RAW. 
I have been using IRScrutinizer which, for example, converts:

Off (Pronto):  
0000 0069 0007 0000 032a 010e 005a 010e 005a 005a 0168 005a 005a 010e 005a 00b4 005a 01c2
to
Off (Raw):
+20518 -6839 +2280 -6839 +2280 -2280 +9119 -2280 +2280 -6839 +2280 -4560 +2280 -11399

Is this the best way to do this? Are there any additional conversions that need to be made given the strange protocol?

To confuse things further - I'm sending with a device based on this: 

https://github.com/markszabo/IRremoteESP8266

Which I believe is based on the Arduino IRremote library.

closed 2020-07-04 by ArminJo